collin80/SavvyCAN #219: UDS scanner stuck

exekutive, 2019-11-13:
There seems to be a problem with the UDS scanner. After it finishes a scan it will keep spamming the bus with messages at a really fast rate. Further scans cannot be performed. Device needs to be reset to make it stop.

[image]

collin80, 2019-11-14:
Hmm, hadn't run into that. I wonder if it's because the lower and upper bound are the same. That is, of course, a valid thing to do but I usually have them spread apart a bit so perhaps that's the issue. Naturally I'll have to better track this down and fix it.

exekutive, 2019-12-01:
still happens if I spread apart the bounds.

[image]

keeps spamming 30 00 03 00...

exekutive, 2021-07-04:
I'm still having the issue where Savvycan starts to bombard the ECU with "30 00 03" when UDS scanner is opened.
It makes it difficult to focus on the data you want when it keeps getting mixed in there.

I tried Mac version, Win 32, and 64.

collin80, 2021-07-04:
It appears that what is happening is a bug in two parts. One: the code is detecting that some traffic coming in looks like ISO-TP encoding for the first frame of a multi-frame message. This is likely not true and some random frames are just triggering it anyway. But, the UDS scan window sets to allow for responding to ISO-TP multi-frame messages so it does indeed respond with a flow control message (0x30 00 03). Two: The UDS scan window should be turning all of that off after it is done scanning and it does not. So, clicking scan just one time turns on the functionality and it never stops because windows in SavvyCAN are never truly closed, just hidden when you "close" them. This allows for rapidly bringing them back when you re-open them but means that things can still technically work in the background even when windows look closed. It may perhaps be a good idea to actually de-allocate the objects for closed windows and disconnect all signals. 

But, the biggest bug here is that it is responding with flow control messages to frames it shouldn't be. This is because the implementation isn't all that great. It should be only responding with flow control if it is really supposed to be the other side of the ISO-TP comm. It doesn't check for that, it just plain responds with flow control to anything that looks like a proper message. I'm sure the implementation could be improved so that it only responds when appropriate.

closed 2022-12-09 by collin80