collin80/SavvyCAN #220: feature request: DBC shortcuts

exekutive, 2019-11-13:
I feel like SavvyCAN is missing a key feature that would make it super efficient. Or perhaps there is a way and I'm just not aware.

Let's say you're monitoring bus traffic, and you spot an ID that you're interested in. I wish I could right-click or double-click (or whatever) right on that frame to start editing a new DBC entry for it.

Also specifying ID ranges in DBC editing.
For example 0x621-0x63F are all virtual networks so it would be handy if one could name all the IDs at once.

Cheers.

collin80, 2019-12-15:
I agree on your first point. For the second part, please explain it a little more. For ranges do you mean you want the ability to create and name a range of IDs (nodes) all at once? That is, cause it to automatically add a postfix to a given name and, say, name them node621, node622, etc?

exekutive, 2019-12-16:
Exactly. And judging from your response, I take it they need unique names. So yes, sequential names is a great idea.

collin80, 2020-07-16:
I should still do this. There is a complication to the request to right click to create a new definition. Currently the hierarchy is Node->Message->Signal. If one were to create a new message definition then the node it is attached to needs to be set too. In the usual work flow this is done by first having a node active then creating a message within that node. So, the hierarchy is obvious. Currently new messages are constructed by clicking on the empty message at the bottom of the list and filling it out. It might be better to create a message editing screen that is used for creating a new message or editing an old one. Such a screen could then explicitly set the sending node to associate it (or leave it at Vector_XXX to not specify one). This also could fix a current oversight - there is no good way to move a message from one node to another if you make a mistake. The editing screen could make this super easy, just select a different node from the drop down list and it's done. This seems like the best solution to make life easier all the way around.

For the second part, I think the best solution is to be able to "clone" an existing message into a new one. Currently this is possible with signals inside of a message. Pressing F2 when a signal is selected clones it with a random number afterward. For messages this same thing could work but increment the number on the end if there is one. So, if a message is named VirtualNet612 and F2 is pressed then the new message is VirtualNet613 if that doesn't already exist.

collin80, 2020-07-30:
There is some progress on this. I totally reworked the DBC GUI and now it is much closer to being able to support right clicking on a message in the main screen to edit or create a message in the DBC file. Of course, since SavvyCAN supports having multiple DBC files loaded this could still be interesting. If two files are loaded which should be used? I suppose I could prompt for the file to add it to.

exekutive, 2020-07-30:
Or the option to create a new one.