collin80/SavvyCAN #222: adding graphs

exekutive, 2019-12-01:
I couldn't figure out how to add a graph. My signal is already defined in a DBC file. The documentation on adding graphs is pretty sparse.

I tried selecting the RPM signal on the right hand side of the dialog and then hit 'add' but nothing happened. I still just see a blank graph. The signal clearly works in the main capture window.

[image]
[image]

collin80, 2019-12-01:
You're right, the documentation could be better (keeping this issue open until I fix that).

You need to select the DBC signal then click "Copy Signal Parameters" to bring them over to the left hand side. Then clicking "Add this graph" will graph that selected signal. The reason for the extra step is that it allows you to edit the signal if perhaps you'd like to scale it or change colors. Or, you can even tweak the signal length and starting bit to see if you can refine the signal somewhat. That's why you have to copy the DBC signal to the graphing definition. But, it could be better explained in the built-in help.

exekutive, 2019-12-01:
Oops. I was updating my original post but then you replied and now I look dumb. You're too quick lol.

Anyway, I should have mentioned that the "Copy Signal Parameters" button wasn't doing anything. I restarted SavvyCAN and now that button works. However, when I hit the "Add this graph" button, nothing happens. I still get a blank graph. I tried restarting again. This is on MacOS.

exekutive, 2019-12-01:
I just noticed that the instructions are in fact there, it just wasn't obvious or accessible from the "Graphing Window" section. The table of contents needs some indenting or divisions.

exekutive, 2019-12-01:
I found the graph by scrolling to the right. It was way off in the rhubarb somewhere. It doesn't seem to be updating though. The reset view, and follow options don't seem to do anything. The graph behaves very strangely and I can't seem to get it to look nice.

collin80, 2019-12-01:
Seems like the OSX build is having trouble. What version of OSX are you using?

exekutive, 2019-12-01:
10.14.6

exekutive, 2019-12-02:
OK I managed to build the app in xcode. Only 889 warnings :-/
Anyway it works a bit better. The signal viewer seems to work. The graph sort of works if I choose 'follow...'. Reset view doesn't work and the window just looks really ugly. No legend or x-axis and vertical label is cut off. (as seen in my screenshot). Arbitrary x offset. Zooming and scrolling still feels very awkward.

exekutive, 2019-12-02:
my binary is 3MB versus your 43MB. ??