collin80/SavvyCAN #223: Signal Viewer no output

exekutive, 2019-12-01:
The signal viewer doesn't seem to work. The value column stays blank.

Also the combo box should be sorted or organised somehow. It can be difficult to find the parameter you want in a very long list.

[image]
[image]

exekutive, 2019-12-01:
update: I closed and reopened signal view and the value is now there. But it's frozen and doesn't live update.

collin80, 2019-12-01:
Hmmm, well, I just checked in Linux and adding signals does seem to work the first try and they do update. So, given this and your other issue I'm starting to suspect that there is a problem with the OSX build. I'll have to check there and find out what's going on with OSX.

exekutive, 2019-12-01:
Do you have a Mac box to test on? I can put a little bit of time into compiling this thing.

collin80, 2019-12-01:
Kind of... I have a virtualbox image of MacOS that I use for testing. Shh.... The actual binary you downloaded was automatically generated by a CI step so I didn't actually compile it. I just did very quick testing to see that it would run. I was naively assuming that code that runs fine on Linux and compiles on MacOS would work the same. But, unfortunately that's not always the case.

exekutive, 2019-12-02:
yeah nothing is ever truly cross platform. Let's see if I have the proper toolchain and how many dependencies I'm missing ...

collin80, 2019-12-02:
Hmm... I fired up my virtual copy of MacOS 10.14 and re-downloaded SavvyCAN from the website so I could be sure I had what everyone else is running. Then I tried loading a Nissan Leaf DBC file and a capture from a 2013 Leaf. I could graph things properly and they'd show up in view and properly scaled. I could load the capture in the playback window then use the signal viewer to select a couple of signals and playing back the log properly shows the signals to be updating and displaying data. I tried the UDS window too but without any capture hardware hooked up it doesn't seem to do a lot. But, it didn't appear to get stuck either. So, at this point I'm striking out on reproducing any of the issues. But, I have heard of other people having issues with MacOS and SavvyCAN. I just haven't found the reason for that. 

Maybe it would help to use real hardware. But, I really am not a fan of Apple or MacOS so I really don't particularly want to purchase a Mac. It's bad enough that I have a hand-me-down iPad. Still, I might have to break down and get something I can run MacOS 10.14 or 10.15 on.

exekutive, 2019-12-02:
Signal viewer seems to work ok in my own build. At least for frame playback. Haven't tried live vehicle connection yet.

closed 2022-12-09 by collin80