collin80/SavvyCAN #224: Building on MacOS

exekutive, 2019-12-02:
I installed the Qt Creator IDE, and used it to compile to see if I would get normally behaving graphs. (created 2.8MB application.)

Results are the same.  Graph is still screwy.

Do you also get around 60 warnings when building?

[image]

collin80, 2019-12-03:
Yes, it has a lot of warnings. A while back I started to squash some of the bugs and got it to have less warnings on Windows and Linux. But, the XCode compiler is super picky and generates warnings where the other platforms do not. Not that they're warning free either. They just generate less.  The reason your binary is so much smaller is that it's using the system installed QT libraries. The binary I distribute has the libraries within the executable folder so that people can use it without needing to install QTCreator.

exekutive, 2019-12-03:
I thought so.

Well that's annoying. I don't know enough about qt to fix this. Using MS Windows makes me want to throw the computer out of a window, and Linux is way too much work for day-to-day use, and I'm not about to go buy a new laptop with either OS. so I guess Mac users are screwed for now.

collin80, 2019-12-15:
I'm trying to install MacOS 10.15 right now. Hopefully I can get it to work in the virtual machine and then find the source of all these weird problems that people are having on MacOS higher than 10.14.0

KostyaSha, 2020-09-26:
Looks like most of then is integers comparison https://stackoverflow.com/a/20543204/4154231

collin80, 2020-09-27:
Yes, most of the warnings are related to comparison between signed and unsigned integers. I'm aware that this isn't a particularly great idea. But, there are many places in the code where QT uses, say, signed integers for something but unsigned integers make more sense for the rest of my code so I use unsigned. And, since those clash you get a warning. The fix would be to figure out whether I can switch to the same integer type that QT wants and if not determine which way to cast to most correctly fix the problem. Of course, it's a lot of busy work that mostly doesn't change a thing so it hasn't been super high priority. But, I do sometimes fix a few of them. So, the number of places where integer comparisons cause warnings should be slowly going down.

KostyaSha, 2020-09-27:
What if move to static helper method that will do the same check along with checking for < 0?