collin80/SavvyCAN #229: Is this device compatible ?

AndKe, 2019-12-21:
https://www.ebay.com/itm/264114364000

Also: 
I have an Saleae 16pro analyzer that can save the data as raw or decoded data... are you aware of any compatibility with Saleae logic software ?

mitchdetailed, 2019-12-21:
Yes, it's capable of utilizing savvycan
Via the Linux 'socketcan' interface option.

AndKe, 2019-12-22:
Great, I take that the injection/replay of frames will work too.
This is very nice, usually, I'd use a microcontroller to do this kind of reverse-engineering, but this seems much faster.
Merry Christmas.

closed 2019-12-22 by AndKe

collin80, 2019-12-24:
Yes, getting access through socketcan will allow for injection and replay. A Saleae Logic 16 will be able to capture data but not send. And, there is no way to integrate Logic with SavvyCAN. Though, it might be an interesting idea to allow SavvyCAN to read Logic captures.

mitchdetailed, 2019-12-24:
Raspberry pi + dual can-fd hat + pimoroni phat is what I'll be doing to
incorporate this. Just have to set event based actions within a script.