collin80/SavvyCAN #230: candump not showing CAN messages published by SavvyCAN

dhoomakethu, 2020-01-31:
Here is what I am trying. Create a fuzzy payload on the bus using SavvyCAN and see how the ECU behaves . I can see the messages on the SavvyCAN but looks like nothing is being sent on the bus. [candump](http://manpages.ubuntu.com/manpages/bionic/man1/candump.1.html) tools no bus activity.

I tried this on a linux machine running ubuntu 18.04 on VirtualCAN as well as PeakCAN.
The bus connection window say it is connected to the bus.

I have tried [latest continuous build](https://travis-ci.org/collin80/SavvyCAN/builds/625385415), [release v199](https://github.com/collin80/SavvyCAN/releases/tag/V199.1) and also built the binary locally as well. All with the same behaviour. 

Please let me know if I am missing something in the entire process?

collin80, 2020-01-31:
Maybe ensure that you've got the proper bus selected. I don't recall if it
automatically sets output to the first bus or not. If not it really should.
Are you using the fuzzing screen or how are you trying to output the
messages?

dhoomakethu, 2020-01-31:
Yes , I am using fuzzing screen and I have only bus which is active at a time either vcan or peak can and the window says it's connected

llvs, 2020-06-30:
With current git master V199, compiled using Qt 5.12.9, I was able to generate traffic via cangen on virtual SocketCAN device vcan0. Also the other way round worked: Playback from SavvyCAN is seen in candump on vcan0. Another SocketCAN-based software I wrote also could exchange messages with SavvyCAN.