collin80/SavvyCAN #231: Error on parsing DBC signals

cyber5tar86, 2020-02-06:
Using your tool i've found a bug on this signal:
 SG_ ACT_PWR : 0|32@1- (1,0) [-3.4E+038|3.4E+038] "kW"  PUMA

it's tagged as SIGNED type, instead it should be a float

collin80, 2020-02-16:
Well, I don't actually see the problem. The thing is, @1 = Little endian integer. For a float it should be either @2 or @3. Though, for a 32 bit single precision float it would be @2. So, it seems SavvyCAN is accepting the definition as presented. If you do actually want it to be a float then change that line to use @2 instead of @1

closed 2020-02-16 by collin80

jackm, 2020-11-07:
> The thing is, @1 = Little endian integer. For a float it should be either @2 or @3. Though, for a 32 bit single precision float it would be @2. So, it seems SavvyCAN is accepting the definition as presented. If you do actually want it to be a float then change that line to use @2 instead of @1

Is this true about byte order values > 1? Where did you come across this information? As far as I am aware, the byte order field can only be `0` (big endian) or `1` (little endian).

Perhaps you meant the `SIG_VALTYPE_` section where 1 => 32bit IEEE float and 2 => 64bit IEEE double? Both must be of signed value type (`-`, not `+`) since they are floats.

Ozuba, 2023-11-03:
Hey i was having a problem related to this issue, while converting the dbc to .c/.h files using cantools the signals defined  in the dbc editor as single precision and double precision dont get translated properly to floats. 

This may be because the way savvycan handles the floating type extending the endianness section options, i think the proper way shoud be using the `SIG_VALTYPE_` section instead.