collin80/SavvyCAN #234: Use with PocketBeagle?

pdp7, 2020-03-31:
@collin80 I was just asked by KC of the Car Hacking Village if the [Macchina P1 (cape for the PocketBeagle)](https://docs.macchina.cc/p1-docs/) would work with SavvyCAN.

The PocketBeagle will show up on the host computer over USB as both a virtual serial port and a network interface.  The host will see the PocketBeagle as:
- /dev/ttyACM0
- 192.168.7.2 (ssh works)

The PocketBeagle only has 1GHz ARM Cortex A8 with 512MB RAM (running Debian armhf). I think it would tough to compile run SavvyCAN directly on it and it would require X11 forwarding as the PocketBeagle has no HDMI.

Do you think there is a way to expose the SocketCAN interfaces on the PocketBeagle (Debian armhf) to the host running SavvyCAN?

collin80, 2020-04-01:
Hmm, that's an interesting question. For comparison, I have compiled SavvyCAN for the RaspberryPi4 and I'm aware of multiple people doing so on the rPI3B+ units. In each case, it worked fine. It's a little bit small if you're using an 800x480 touchscreen on the rPi but it works. So, conceptually the idea of running it on smaller embedded hardware isn't so far fetched. SavvyCAN will use socketcan devices on linux so I think the answer is that compiling it and running it would expose the socketcan interfaces for use. You're right, it might be rather tough to make it work on a pocketbeagle since it has no native video. It is possible that maybe X11 forwarding or virtual VNC could work. It appears that QT does support embedded linux and that people have compiled QT programs on a Beaglebone board. So, while difficult, it does seem possible.

Entropy512, 2020-05-05:
@pdp7 - Do you want to interact with it in real time using SavvyCAN, or just view logs?

At least in the interim, you could log on the P1, and open the logs in SavvyCAN.

I have a P1 on order myself.  Hoping it gets here by the weekend, but unlikely.

pdp7, 2020-05-09:
@Entropy512 thanks, copying the logs from the Beagle might be sufficient (it was someone else who was asking me so I'll have to check on their use case)