collin80/SavvyCAN #236: Idea for a new function within your program

RawLiquid, 2020-04-09:
Greetings, I have been following this project for a very long time, and have been trying to add a feature for some time with no success at even figuring out how/where to start. The intended purpose would be to assist with locating specific values, where you know what you want, approximately what data size your looking for, and in theory know how the value should be changing, regardless of how quickly or how often it is being sent. It is almost a combination of the sniffer window, but with a similar function to the flow view.
You start off with a reference snapshot of all pid's and the last data frame for each, usually by letting things run for a minute before initiating the function. When you start the function, it should ask what the smallest datatype you think the value could be, and whether or not it is a value that you readily have access to and can input. After a short amount of time, and doing something that should have caused the target value to change, you press one of 3 buttons, -=+, basically telling savycan to save the current snapshot, compare it to the initial reference data and using the datasize given earlier, discard all pids which did not change in the specified way. namely - for went down, = for stayed the same, and + for went up. you then repeat that process, each time reducing the possible places your target value could be. A secondary, more cpu intensive analysis could be performed not in realtime, which could include larger datatypes, and/or include known common formulas normally used to obfuscate different values.

If my description was a bit confusing, head over to youtube and search for 'cheat engine walkthrough' and you can see the concept used to do the same thing, but for video games. I would love to help, but figuring out how to do anything in the qt framework has been rather frustrating to say the least.

collin80, 2020-04-10:
It's not a bad idea. I tried to do something similar with the range state window which is hidden in the binary builds but available if you enable it from the source code. But, I was never happy with my design of that window and never felt like I had a good enough plan for how to execute it. Your ideas will be beneficial in restarting that whole thing. I would like something like this to be available in the program. So, thanks for the suggestions.

kdschlosser, 2021-02-18:
This is a mechanism I used to pinpoint what I needed when reverse engineering.
I used this approach when reverse engineering the CAN traffic on my vehicle. 

This would be for reverse engineering data frame that get created by a user action. Like pressing the door lock button, or turning on and off traction control those kinds of things.

When the user presses the start button a timer begins and the program captures frames for 10 seconds. then it instructs the user to to do whatever it is they need to do to cause the frame to get transmitted once they press the OK button. and once they have carried out the action to press the Done button. the program records all the frames receiver between the pressing of the OK and the Done buttons. it looks at the second capture and if there are any frame id's in it that are also in the first capture it removes those frames from the second capture it also removes any frames that are duplicate and the data has not changed. then it does the same thing as last time except it tells the user to toggle/carry out the same action 10 times... when finished the program removes frames that are in the first capture and it removes frames that are not in the second capture. then it looks for frame ID's that have been repeated 10 times where the data has not changed and also looks for 20 frames where the data has "toggled". it then presents the found information to the user. and also the ability to pick a frame and transmit it on the bus to see if it does what they want it to do. 

This is a good way to locate frames for various systems in a vehicle that are generated because of what a person does. It can also be used to locate sensor data by unplugging the sensor and plugging it back in.