collin80/SavvyCAN #237: pcapng (Wireshark/tcpdump) capture support

Entropy512, 2020-05-05:
A nice thing to have in SavvyCAN would be Wireshark/tcpdump pcapng file support.
Wireshark/tcpdump can capture from SocketCAN interfaces.
If desired, I can provide some example pcapng files with single-bus CAN data, and once my Macchina P1 comes in, hopefully multi-bus examples.
I also have some example Python code that shows how to work with these files and python-pcapng https://pypi.org/project/python-pcapng/ which is probably better than the pcapng specs as far as certain idiosyncracies of pcapng CAN traces if desired.

(Alternatively, any suggestions for a conversion tool that will output in a format SavvyCAN understands would be useful, so I don't have to write one myself.  :) )

collin80, 2020-05-06:
Yes, that would be interesting to add. I'm adding this to the list of things I should add to the software.

Entropy512, 2020-05-06:
I'll try to get some documentation on the CAN-specific quirks of pcapng that I've found put somewhere publically by this weekend.  Specifically, if you do a capture on newer kernels vs. older, the exact encapsulation is a little different.

I'm not sure if it's worth messing with the older formats at this point, given that "newer" goes back to the kernel used by Ubuntu 18.04 at least.  It changed somewhere between 16.04 and 18.04

collin80, 2022-12-09:
Someone did add support for wireshark captures. But, now I don't remember the exact hash to reference for which commit that was.

closed 2022-12-09 by collin80