collin80/SavvyCAN #238: CANtact support

romain145, 2020-08-11:
Is there a plan to support the CANtact / CANable / CANable Pro interfaces running the stock slcan firmware (serial) or the candleLight firmware?

collin80, 2020-08-11:
SavvyCAN supports socketcan so if you're running candlelight then you can
use socketcan. Slcan should work as well as that's also socketcan.

romain145, 2020-08-12:
Using the candleLight firmware (flashed here https://canable.io/updater/) the port list is empty.
I tried passthrucan, peakcan, systeccan, tinycan (shows a can0.0 anyway) and vectorcan.

Testing on Windows 10 btw.

rusoku, 2020-08-12:
RUSOKU TouCAN USB CAN converter driver are based on Qt CAN bus plugin so I quess it should work under Windows.

mitchdetailed, 2020-08-12:
Romain, use Linux in a VM with the Candlelight firmware + canable if you're
needing a Windows based solution..

Candlelight will work with socketcan natively without using slcand
commands. (sudo ip link set up can0 type can bitrate #######)

Then pull up the needed interface in Linux VM on windows.

collin80, 2021-08-29:
Hmmm, interesting, yes that does appear to be a Windows driver for the candlelight API. I will look into supporting that.

BennyEvans, 2022-08-02:
I realise this is an old issue, but has there been any further investigation on supporting these devices in SavvyCAN under Windows?

collin80, 2022-08-03:
I never did look into it. I just went to that link and bookmarked it. I'll see what I can do.

ElektronikCZ, 2022-11-16:
Windows has candlelight support out of the box with no driver install needed. I also give +1 to this request, and im hoping you atleast looked into it. Its annoying to flash a different firmware in my can tool just for savvycan

Alia5, 2022-11-17:
Meanwhile:

WSL2 + Custom kernel (to include can-bus stuff) + [usbipd-win](https://github.com/dorssel/usbipd-win).

And then just use savvycan on the wsl

gromgsxr, 2023-01-14:
is support likely to get added for this?

collin80, 2023-01-15:
Someday, probably. But, as covered above, CANTact support works fine through the socketcan interface on Linux so using WSL could work. It's certainly not ideal. I really should implement actual support in Windows one of these days.

mitchdetailed, 2023-01-15:
could always port the pcan code to the cantact.

gadjex, 2023-01-20:
Came here from Google search looking for the same thing, support for CANable in Windows. Would be nice if this was done.

gromgsxr, 2023-01-20:
> Someday, probably. But, as covered above, CANTact support works fine through the socketcan interface on Linux so using WSL could work. It's certainly not ideal. I really should implement actual support in Windows one of these days.

Direct windows support would be just awesome

Careless, 2023-01-24:
> Someday, probably. But, as covered above, CANTact support works fine through the socketcan interface on Linux so using WSL could work. It's certainly not ideal. I really should implement actual support in Windows one of these days.

this doesn't help macOS users, though.

ElektronikCZ, 2023-03-01:
> Someday, probably. But, as covered above, CANTact support works fine through the socketcan interface on Linux so using WSL could work. It's certainly not ideal. I really should implement actual support in Windows one of these days.

You need to compile your own WSL kernel for that to work, and i never got that running. The SLCAN serial FW doesnt allow you to set different can bus speeds, like 95kbps as far as im aware. So the only option for me seems to be to install a full linux VM.
There is a cangaroo software which you could use as the basis for porting candlelight support to windows
https://github.com/HubertD/cangaroo

BennyEvans, 2024-05-02:
I've got some spare time now so will look at implementing this. I'll create a PR once complete and link back here.

ElektronikCZ, 2024-05-02:
This is one of the most epic comebacks of 2024

BennyEvans, 2024-05-13:
I've had a chance to get some initial code working. If anybody in this thread has a chance, it'd be great if you could also test. I would like to do some more testing before I create the PR but will hopefully have a PR up in the next month. This work was also ontop of another modification that adds support for opening Wireshark socketCAN files. I'll create a PR for that first.

**There are currently some restrictions with the code:**
* Only supports devices with base clocks of 48Mhz or 80Mhz - these are the clocks of the two devices I have (CANable and the device I'm currently developing, CAN Bus Debugger). If someone can send me the capability of the CANtact, that can be added pretty easily too if it's not one of the above.
* Does not yet support one shot mode etc. but does support listen only.
* Currently only compiled in when using MinGW to compile on Windows - I'm not sure it'll work so well if compiling under msvc but if someone has a chance, please feel free to modify the .pro file and include support when compiling with msvc. Let me know how you go.

**I still need to test:**
* Multiple devices connected at once
* More TX testing
* Test compilation on linux/mac

Also please note that candle_dll is licensed under the LGPL. This is fine, as it's compiled separately and dynamically linked. But if you're wondering why the driver isn't just compiled into SavvyCAN, that's the reason. Any final Windows release containing the dll will need to contain a note regarding the dll license and where the code can be found - but this is trivial.

Also for transparency, please note that I'm the developer of a new product (not yet released) called CAN Bus Debugger.

Code is available here; https://github.com/BennyEvans/SavvyCAN/tree/gs_usb_support
The code to compile the dll is here: https://github.com/BennyEvans/candle_dll

BennyEvans, 2024-05-29:
Update on this: I've been away but plan to get back to testing soon. I've just creted PR #790 to add support for SocketCAN PCAP files and once that PR has been merged and I've finished testing I'll create the PR for gs_usb (candle) support.

BennyEvans, 2024-07-10:
PR #790 has just been merged, so I'll work towards creating the next PR to add support for gs_usb (candle). I have something working, but would like to tweak a couple things and as I now have two devices, would like to ensure it can operate with multiple devices attached.

BennyEvans, 2024-08-27:
I've just created the next PR #835 that adds support for these devices on Windows. Hopefully we can soon mark this issue as complete 🎉