collin80/SavvyCAN #240: How to add Firmware upload view

mattisalo, 2020-09-01:
Bit new to these sources... How to get Firmware uploader to be visible on windows version?

collin80, 2020-09-01:
If you're using the WIP branch then these lines start at 192 in MainWindow.cpp:

    //these either are unfinished/not working or are not for general use. But,they exist
    //so if you want to enable them and play with them then go for it.
    ui->actionFirmware_Update->setVisible(false);
    ui->actionMotorControlConfig->setVisible(false);
    ui->actionSingle_Multi_State_2->setVisible(false);

Change whichever you want to true and recompile then you'll see that window as an option. I haven't tried the firmware updater in quite a while. Bit rot might have gotten to it.

mattisalo, 2020-09-10:
THX ,i got it enabled. But when i push the "start upload" nothing happens. I'm trying to use PCan as canbus adapter. No can messages are sent. Bo something really has happened to it....

collin80, 2020-09-13:
Yeah, sounds like it is broken. I recently re-factored the whole CAN bus system and that probably broke the firmware uploader window. I don't ever use it anymore so it wasn't a big priority for me to get it fixed. In fact, I don't know if I even have any hardware laying around to test with at the moment. But, I'll put it on the list of things to fix.