collin80/SavvyCAN #241: Found solution to QT rendering issue in Linux Mint

jerkey, 2020-09-08:
I run Linux Mint and SavvyCAN has never rendered properly and required a lot of tweaking of field widths (which are not persistent).  Today I found that the following environment variables make it work great!

export QT_SCALE_FACTOR=1
export QT_AUTO_SCREEN_SCALE_FACTOR=0
export QT_SCREEN_SCALE_FACTORS=2

I found them at this link:
https://www.reddit.com/r/linuxmint/comments/eca468/qt_apps_hidpi_scaling_in_cinnamon/

I am not sure which of the three fixed the issue, and I don't know if this is more of a Linux Mint issue or a SavvyCAN issue, but I wanted you to know in case there's something you can do so that other people don't have to go through the same process.

I have a 3200x1800 display, which is "HiDPI" and I think that affects things, but anyway the above worked great.  Here's a picture of extended CAN frames before (need to scroll sideways to see them!?!?) and then after the above fix:
[image]
[image]

KostyaSha, 2020-09-26:
HIDPI on osx also produces strange result with width, height, centering...

collin80, 2020-09-27:
It has been a while since I last tested on a HiDPI device. I've got a 4K monitor but I rarely use it. I'll have to work on this some more.