collin80/SavvyCAN #244: Move interpretation into separate column

KostyaSha, 2020-09-28:
I think it would make sense:
1. to move DBC intepretation into separate column (and add/hide it with existing parsing checkbox)
2. add interperation for sniffer
Would you accept such kind of improvements?
Also, does it look the same for everyone? vertical alignment, extra spaces?
[image]

collin80, 2020-09-28:
Back when I added interpretation I thought about using a separate column for that. I believe I decided against it just because it would increase the amount of horizontal space required. By putting it in the same column it trades some horizontal space for vertical space. Since messages tend to have many signals you end up with a lot of vertical space taken up anyway. So, it seemed a fair compromise. However, I never tried a separate column to see how it looks. Perhaps you could try it and see how you like it. If it looks OK then, sure, send a PR and I'll take a look too. I suppose interpretation could be added to the sniffer window too but it seems like that window is already a bit busy. This is another place where I'd want to see what it looks like before saying whether I like it or not.

KostyaSha, 2020-09-28:
Also on screenshot colorstrip is lost..

In [my tool](https://github.com/KostyaSha/bean-cracker)(opensourced right now, it's heavy draft) i spent a lot of time making compact and clean tables data (either it not comfortable to deal with data). 
[image]
Will try some changes on SavvyCan and sent PR with screenshots.