collin80/SavvyCAN #251: Python scripting?

KostyaSha, 2020-10-05:
Searching the way to make gateway with packets modification. Scripting looks interesting. 
Just curious why JS? WDYT about python? Would it be difficult to add bindings?
Checked also network for candevstudio and found https://github.com/GENIVI/CANdevStudio/compare/master...rkollataj:pyscripter2

collin80, 2020-10-05:
Javascript was really easy to integrate. That's the only reason. It was simple to do. QT has built-in JS scripting support so I could create simple bindings and be done with it. It technically works but nobody wants to use JS for CAN scripting so basically no one uses the scripting interface. You're right, Python would be more to everyone's liking. Alas, it hasn't shown to be quite as simple to create a cross platform program that allows for Python scripting. But, CANDevStudio runs on all the same platforms as SavvyCAN so their python scripting must work across all three platforms. I've investigated various options for python embedding but none of them have seemed nearly as easy to integrate and get working on Windows/Linux/Macos. I'll look more deeply at pyscripter2 in CDS and see if it gives me some ideas.

Totally off topic:
Funny enough, I was JUST using CDS myself. But, the DBC loader in CDS is, sorry to say, total garbage. It won't load any DBC files I try to load. So, that's kind of sad. I was thinking of adding to SavvyCAN a way to send messages/signals from a DBC file. Then I remembered CANDevStudio and thought I might just tell people to use that instead. But, not if it can't load even reasonable DBC files that everything else loads perfectly well.

KostyaSha, 2020-10-06:
I thought python was something easy to integrate because i saw it everywhere near Qt, thanks for details. Better to have JS than nothing of course. 

as off topic:
I wanted to wrote qt based can tool, but hopefully found this project when started searching DBC libraries. CDS looks interesting but i was unable to do simple sends... One of the openpilot DBCs silently didn't load. In savvycan i firstly want to make UI more compact, qt by default ate a lot of monitor space and thinking to try qtwidgets as PoC. When you have a lot of windows it's pretty difficult to get in right order focus on them. Probably with qwidgets it would be possible to have tabs and some tile region where any tool could be placed...

closed 2020-10-06 by KostyaSha