collin80/SavvyCAN #253: Support non can packets for transforming into can

KostyaSha, 2020-10-07:
Question. I have BEAN Bus packets that didn't exactly match to can data and i want check bean packets data and transform into CAN packet and send. That something that i will do with microcontroller, but I would like to draft and develop data with scripting fist. Could you guide what i can try adopt/modify? Probably i need bind some function into JS and create new source of packets (may be i can refactor my Qt app and make shared code with connections  window).

mitchdetailed, 2020-10-07:
Hey Kanstantsin, I'd personally recommend running the packets through
python-can to import or use them. Set them up as a vcan0 transmitted
message when received, that's bridged to can0

Nonetheless - I encourage you to persevere, as the Toyota/Lexus community
doesn't have much aftermarket input regarding Toyota's BEAN protocol

KostyaSha, 2020-10-07:
I got an idea, temporary i can map msgid (seems it's unique) to can id and data to data and send to some virtual savycan interface (mqtt is already here?).
[image]

> Nonetheless - I encourage you to persevere, as the Toyota/Lexus community doesn't have much aftermarket input regarding Toyota's BEAN protocol

Protocol is pretty simple :) There is also avc-lan...