collin80/SavvyCAN #254: Connection window confusion

KostyaSha, 2020-10-11:
It seems that reset device is deleting device. What is the idea of this buttons?
As user i would expect add/delete  connect/disconnect 
```
    void handleRemoveConn();
    void handleNewConn();
    void handleResetConn();
```
[image]

KostyaSha, 2020-10-11:
There is FROM `connectionwindow.ui` and `uic` tool should generate corresponding `ui_connectionwindow.h`, but it looks like manually managed. Is it true? How do you usually do UI changes?

collin80, 2020-10-11:
I'm not aware of anything in the UI generation being manually managed. As you said, uic should generate all the files automatically. I have found that sometimes I'll end up with a bunch of files stuck in my source folder that override the automatically generated copies. I don't know why that happens sometimes but it shouldn't. If it ever does you need to delete the copies from the source folders or it will preferentially use them and that's a real pain. That has happened to me a couple of times and it becomes really confusing when you make UI changes and... nothing happens. But, it seems to be a QT build glitch that dumps the auto-generated files into the source folder instead of the build folder.

KostyaSha, 2020-10-11:
Regenerated manually. I tried to add more buttons logically to what device can do. 
Does start/stop mean connect/disconnect?

collin80, 2020-10-11:
Yes, Start/Stop in this case does basically mean connect or disconnect. The idea for reset was to basically do a disconnect / reconnect cycle. I don't know that it ever actually got coded properly.

KostyaSha, 2020-10-15:
So when start is not called on creation deadlock happens later. Something about threading...