collin80/SavvyCAN #255: QT 5.13 is minimum required version now

Diginomicon, 2020-10-12:
Building on a Raspberry Pi 4 Using Buster, the pre-build QT5 packages are 5.11, but the build fails due to the introduction of swapItemsAt in commit fb58fc94ddc2f74c952af91e61e35162bcf53845.  swapItemsAt was introduced in QT 5.13, so anyone wanting to build SavvyCAN on a RPi4 will need to 1st built and a newer version of QT5 instead of using the version from repositories.

collin80, 2020-10-13:
Thanks for the update. I remember adding that line because it was far simpler than other approaches. Apparently I was already on QT 5.13 on my machine so it was available and I didn't notice that it was so new. Perhaps there's a reasonable alternative that would be available on earlier versions of QT?

collin80, 2020-10-13:
Apparently I used swapItemsAt because swap (which is basically the same thing) was deprecated. So, you can just change that line to say swap instead of swapItemsAt and it should work fine on older QT versions. Personally I find it a bit silly to rename functions and deprecate the old name.

Diginomicon, 2020-10-13:
I don't know why it didn't occur to me to try to switch back to the older function since their signatures were the same, I spent most of the day getting QT 5.15 to compile and then struggling with it not wanting to run in anything other than full screen mode.  After removing the compiled version and reverting back to the 5.11 available through the Raspberry OS repos, editing line 1755 in dbc/dbchandler.ccp back to just .swap it compiles cleanly and runs fine.

Basic steps to have a functioning copy of SavvyCAN on RPI 4, without having to custom compile QT5:

1. sudo apt install build-essential qt5-default libqt5serialbus5-dev qtdeclarative5-dev qttools5-dev libqt5serialport5-dev
2. git clone https://github.com/collin80/SavvyCAN.git
3. cd SavvyCAN
4. nano dbc/dbchandler.cpp
5. qmake
6. make
7. sudo make install
8 ./install

jackm, 2020-11-03:
Just a note that for Ubuntu 20.04.1 (focal), the [`qt5-default` package version is at 5.12.8](https://packages.ubuntu.com/focal/qt5-default) in the official Ubuntu apt repository so if you needed Qt >=5.13 you would have to manually download and install a newer QT version as suggested in the SavvyCAN README.

Personally I just followed the [steps that Diginomicon specified](https://github.com/collin80/SavvyCAN/issues/255#issuecomment-707461500) and compiled with Qt 5.12.8.

closed 2020-11-13 by collin80