collin80/SavvyCAN #256: Graphing glitches to 0 and sometimes 2^16 with no such messages appearing in a candump

willemcvu, 2020-10-12:
Setup: 

1. Broadcast a constant 0x13 0x88 on ID 0x007 from a piece of hardware I'm debugging.
2. Use SavvCAN to plot ID 0x007, see glitches to zero and occasionally to 2^16.
3. Use candump, see no glitches to zero.

No idea what's going on here. I'm probably being an idiot but I don't see how. This issue has been replicated across 3x computers running savvycan, connected to the same bus.

[image]

[image]

```
willem@willem-XPS-15-7590:~$ candump can0,007:7FF -t d
 (000.000000)  can0  007   [8]  13 88 00 00 00 00 95 74
 (000.009984)  can0  007   [8]  13 88 00 00 00 00 95 75
 (000.009917)  can0  007   [8]  13 88 00 00 00 00 95 76
 (000.010035)  can0  007   [8]  13 88 00 00 00 00 95 77
 (000.009927)  can0  007   [8]  13 88 00 00 00 00 95 78
 (000.009986)  can0  007   [8]  13 88 00 00 00 00 95 79
 (000.009966)  can0  007   [8]  13 88 00 00 00 00 95 7A
 (000.009984)  can0  007   [8]  13 88 00 00 00 00 95 7B
 (000.010011)  can0  007   [8]  13 88 00 00 00 00 95 7C
 (000.009744)  can0  007   [8]  13 88 00 00 00 00 95 7D
 (000.009995)  can0  007   [8]  13 88 00 00 00 00 95 7E
 (000.010011)  can0  007   [8]  13 88 00 00 00 00 95 7F
 (000.009990)  can0  007   [8]  13 88 00 00 00 00 95 80
 (000.009994)  can0  007   [8]  13 88 00 00 00 00 95 81
 (000.009798)  can0  007   [8]  13 88 00 00 00 00 95 82
 (000.009976)  can0  007   [8]  13 88 00 00 00 00 95 83
 (000.010000)  can0  007   [8]  13 88 00 00 00 00 95 84
 (000.010004)  can0  007   [8]  13 88 00 00 00 00 95 85
 (000.009996)  can0  007   [8]  13 88 00 00 00 00 95 86
 (000.010049)  can0  007   [8]  13 88 00 00 00 00 95 87
 (000.010002)  can0  007   [8]  13 88 00 00 00 00 95 88
 (000.009970)  can0  007   [8]  13 88 00 00 00 00 95 89
 (000.009808)  can0  007   [8]  13 88 00 00 00 00 95 8A
 (000.009973)  can0  007   [8]  13 88 00 00 00 00 95 8B
 (000.010074)  can0  007   [8]  13 88 00 00 00 00 95 8C
 (000.009932)  can0  007   [8]  13 88 00 00 00 00 95 8D
 (000.010062)  can0  007   [8]  13 88 00 00 00 00 95 8E
 (000.009822)  can0  007   [8]  13 88 00 00 00 00 95 8F
 (000.009947)  can0  007   [8]  13 88 00 00 00 00 95 90
 (000.009985)  can0  007   [8]  13 88 00 00 00 00 95 91
 (000.010011)  can0  007   [8]  13 88 00 00 00 00 95 92
 (000.010000)  can0  007   [8]  13 88 00 00 00 00 95 93
 (000.009748)  can0  007   [8]  13 88 00 00 00 00 95 94
 (000.009999)  can0  007   [8]  13 88 00 00 00 00 95 95
 (000.009945)  can0  007   [8]  13 88 00 00 00 00 95 96
 (000.010059)  can0  007   [8]  13 88 00 00 00 00 95 97
 (000.010002)  can0  007   [8]  13 88 00 00 00 00 95 98
 (000.010014)  can0  007   [8]  13 88 00 00 00 00 95 99
 (000.009713)  can0  007   [8]  13 88 00 00 00 00 95 9A
 (000.009944)  can0  007   [8]  13 88 00 00 00 00 95 9B
 (000.009979)  can0  007   [8]  13 88 00 00 00 00 95 9C
 (000.010098)  can0  007   [8]  13 88 00 00 00 00 95 9D
 (000.010088)  can0  007   [8]  13 88 00 00 00 00 95 9E
 (000.009790)  can0  007   [8]  13 88 00 00 00 00 95 9F
 (000.010061)  can0  007   [8]  13 88 00 00 00 00 95 A0
 (000.009847)  can0  007   [8]  13 88 00 00 00 00 95 A1
 (000.009853)  can0  007   [8]  13 88 00 00 00 00 95 A2
 (000.010023)  can0  007   [8]  13 88 00 00 00 00 95 A3
 (000.009948)  can0  007   [8]  13 88 00 00 00 00 95 A4
 (000.009904)  can0  007   [8]  13 88 00 00 00 00 95 A5
 (000.010004)  can0  007   [8]  13 88 00 00 00 00 95 A6
 (000.009989)  can0  007   [8]  13 88 00 00 00 00 95 A7
 (000.009909)  can0  007   [8]  13 88 00 00 00 00 95 A8
 (000.009951)  can0  007   [8]  13 88 00 00 00 00 95 A9
 (000.010013)  can0  007   [8]  13 88 00 00 00 00 95 AA
 (000.009975)  can0  007   [8]  13 88 00 00 00 00 95 AB
[... 373 lines cut ...]
 (000.009793)  can0  007   [8]  13 88 00 00 00 00 97 21
 (000.009992)  can0  007   [8]  13 88 00 00 00 00 97 22
 (000.009988)  can0  007   [8]  13 88 00 00 00 00 97 23
 (000.010000)  can0  007   [8]  13 88 00 00 00 00 97 24
 (000.009999)  can0  007   [8]  13 88 00 00 00 00 97 25
 (000.010038)  can0  007   [8]  13 88 00 00 00 00 97 26
 (000.009738)  can0  007   [8]  13 88 00 00 00 00 97 27
 (000.010011)  can0  007   [8]  13 88 00 00 00 00 97 28
 (000.009979)  can0  007   [8]  13 88 00 00 00 00 97 29
 (000.009999)  can0  007   [8]  13 88 00 00 00 00 97 2A
 (000.009998)  can0  007   [8]  13 88 00 00 00 00 97 2B
 (000.010000)  can0  007   [8]  13 88 00 00 00 00 97 2C
 (000.010036)  can0  007   [8]  13 88 00 00 00 00 97 2D
 (000.009766)  can0  007   [8]  13 88 00 00 00 00 97 2E
 (000.009913)  can0  007   [8]  13 88 00 00 00 00 97 2F
 (000.009902)  can0  007   [8]  13 88 00 00 00 00 97 30
 (000.010248)  can0  007   [8]  13 88 00 00 00 00 97 31
 (000.009918)  can0  007   [8]  13 88 00 00 00 00 97 32
 (000.009836)  can0  007   [8]  13 88 00 00 00 00 97 33
 (000.009945)  can0  007   [8]  13 88 00 00 00 00 97 34
 (000.009928)  can0  007   [8]  13 88 00 00 00 00 97 35
 (000.010074)  can0  007   [8]  13 88 00 00 00 00 97 36
 (000.010031)  can0  007   [8]  13 88 00 00 00 00 97 37
 (000.010011)  can0  007   [8]  13 88 00 00 00 00 97 38
 (000.009937)  can0  007   [8]  13 88 00 00 00 00 97 39

collin80, 2020-10-13:
I copied your log data you posted and set up a graph like you showed. When doing that I get a perfectly straight line. So, it seems that loading data from a saved log causes everything to work. So, the next thing to do is see what SavvyCAN is capturing itself. Can you do the same thing but now save a CAN log file from directly in SavvyCAN? I'd be interested to see if the capture saved from within SavvyCAN has corrupted bytes for the first and second bytes. That would explain the graph but not why it's doing that. I assume you're using LINUX and capturing with a socketcan device? Also, which version of SavvyCAN are you using? I'm testing with the bleeding edge continuous builds (which you'll find in the releases for SavvyCAN)

willemcvu, 2020-10-13:
Here's a pastebin from the SavvyCAN log: https://paste.ubuntu.com/p/W88pYGFXyC/

Looks like the data that SavvyCAN captures has the dropouts.

I'm using Linux Mint 20, a socketcan device (Canable), and I've seen this issue on `SavvyCAN-b3a472c` , `SavvyCAN-4573274` and `SavvyCAN-dbccaba` (bleeding edge continous).

collin80, 2020-10-13:
Yes, I'm seeing what you're talking about. I believe that what might be happening is that SavvyCAN is not properly interpreting error frames in the communications. The socketcan driver in SavvyCAN does know about error frames but I'm seeing that we left them going through as if they were honest to goodness data frames. They aren't and shouldn't be directly treated as such. I see that the handling of error frames was commented out which only made matters worse. So, I'm fixing that all up right now. I'm going to support the proper socketcan error interface as much as possible. What is going to happen is that error frames will be marked as an impossibly high CAN ID - starting at 0x20000000 which is just past the largest 29 bit ID you can send. That way error frames will show up in the log but they won't have an ID you could mistake for real traffic. This lets you see when errors happen but not have garbage data in your normal traffic IDs.

collin80, 2020-10-14:
I just pushed #2749c55 to WIP. I think it will interpret errors on the socketcan interface better. But, right now I'm not set up to test this very well so YMMV. Soon the continuous build should complete and you can try it.

willemcvu, 2020-10-19:
All makes sense, I will test when I get a chance. Thanks for figuring this out!

closed 2021-04-04 by collin80