collin80/SavvyCAN #260: Random Segmentation Fault v200

voltlog, 2020-11-18:
Because I was having a bunch of random segmentation fault crashes with the precompiled release, I grabbed QT 5.14.2 and compiled the latest version of SavvyCAN (v200) on Ubuntu 20.04. It still crashes randomly (and often) while using it to send custom frames. I am a noob regarding QT but I ran SavvyCAN in a debug session with QT Creator and this is the info it shows when it crashes. If there is anything else I can provide for identifying the issue here, please let me know. 
[image]

collin80, 2020-11-19:
That's really odd. The line it is faulting on is not the sort of line I'd expect to be able to cause a segfault. The only way I could see that happening is if the pointer to the object ("this" implicit pointer for the object) was simultaneously not null but also not within legitimate memory space. That sounds like the object reference is being set in a faulty manner. It might help to see what is farther down in the stack trace. I can only see the first 10 in your screenshot. It might also help if I saw the exact settings you're using in your custom senders so I could reproduce it the same way you are. Also, this crash seems to be within code that processes multiplexed frames so it probably requires that the custom sent message be within a multiplexed message.

I'm going to try to create some custom CAN frames myself with guesses to see if I can get it to do it. I have DBC files with multiplexed messages so I can load one of those then try to send custom messages on that ID. Truth be told, I never use the custom sender so it might have long standing bugs that I never caught. I'll try to make it crash for me.

voltlog, 2020-11-20:
I've managed to isolate a scenario where it happens, this one is related to DBC interpreting of received signals(no sending during this scenario). I have a couple of incoming messages, "Overwrite mode" selected and a DBC filter file that I created myself([test_dbc.txt](https://github.com/collin80/SavvyCAN/files/5571558/test_dbc.txt)). The moment I activate the option "Interpret Frames" it crashes, full stack trace below.

[image]

> 1  DBC_SIGNAL::processAsInt                                                                                                                               dbc_classes.cpp   144 0x555555713763 
> 2  DBC_SIGNAL::processAsText                                                                                                                              dbc_classes.cpp   64  0x555555713112 
> 3  CANFrameModel::data                                                                                                                                    canframemodel.cpp 485 0x5555555ccafc 
> 4  QSortFilterProxyModel::data(QModelIndex const&, int) const                                                                                                                   0x7ffff58093f9 
> 5  QStyledItemDelegate::initStyleOption(QStyleOptionViewItem *, QModelIndex const&) const                                                                                       0x7ffff78c0500 
> 6  QStyledItemDelegate::paint(QPainter *, QStyleOptionViewItem const&, QModelIndex const&) const                                                                                0x7ffff78bf739 
> 7  QTableViewPrivate::drawCell(QPainter *, QStyleOptionViewItem const&, QModelIndex const&)                                                                                     0x7ffff78e6b8a 
> 8  QTableView::paintEvent(QPaintEvent *)                                                                                                                                        0x7ffff78ef350 
> 9  QWidget::event(QEvent *)                                                                                                                                                     0x7ffff7699670 
> 10 QFrame::event(QEvent *)                                                                                                                                                      0x7ffff774179e 
> 11 QAbstractItemView::viewportEvent(QEvent *)                                                                                                                                   0x7ffff78a3fbc 
> 12 QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject *, QEvent *)                                                                                                  0x7ffff583d9dd 
> 13 QApplicationPrivate::notify_helper(QObject *, QEvent *)                                                                                                                      0x7ffff765aae2 
> 14 QApplication::notify(QObject *, QEvent *)                                                                                                                                    0x7ffff7661a90 
> 15 QCoreApplication::notifyInternal2(QObject *, QEvent *)                                                                                                                       0x7ffff583dc48 
> 16 QWidgetPrivate::sendPaintEvent(QRegion const&)                                                                                                                               0x7ffff76920c5 
> 17 QWidgetPrivate::drawWidget(QPaintDevice *, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter *, QWidgetRepaintManager *)                       0x7ffff7692bc7 
> 18 QWidgetRepaintManager::paintAndFlush()                                                                                                                                       0x7ffff766af03 
> 19 QWidgetRepaintManager::sync()                                                                                                                                                0x7ffff766b648 
> 20 QWidgetPrivate::syncBackingStore()                                                                                                                                           0x7ffff769374f 
> 21 QWidget::event(QEvent *)                                                                                                                                                     0x7ffff76997f0 
> 22 QMainWindow::event(QEvent *)                                                                                                                                                 0x7ffff77a0791 
> 23 QApplicationPrivate::notify_helper(QObject *, QEvent *)                                                                                                                      0x7ffff765ab0c 
> 24 QApplication::notify(QObject *, QEvent *)                                                                                                                                    0x7ffff7661a90 
> 25 QCoreApplication::notifyInternal2(QObject *, QEvent *)                                                                                                                       0x7ffff583dc48 
> 26 QWidgetRepaintManager::sendUpdateRequest(QWidget *, QWidgetRepaintManager::UpdateTime) [clone .part.94] [clone .constprop.96]                                                0x7ffff766c4b6 
[... cut ...]
> 50 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)                                                                                                                      0x7ffff583c662 
> 51 QCoreApplication::exec()                                                                                                                                                     0x7ffff5845590 
> 52 main                                                                                                                                                   main.cpp          24  0x55555559a598

collin80, 2021-05-23:
I've finally gotten some time to look at this. While I'm looking at the current version and not v200 I see a potential problem with your DBC file. You have RPM defined as a multiplexed signal but there is no multiplexor to use with it. This actually immediately crashes the current version when you try to load it. I'm going to fix loading of such files to at least not crash. Basically, if there is no multiplexor it'll have to revert to non-multiplexed mode which is perhaps what you wanted anyway. I think the previous version was allowing the DBC to load but if a signal was marked multiplexed but there was no multiplexor to check for value it would crash.