collin80/SavvyCAN #267: Error in dbc/dbchandler when make

tsunemitsu711, 2020-12-16:
In dbc/dhchandler.cpp, line 1755, error: 'class QList<DBCFile>' has no member named 'swapItemsAt '　
I get an error like this and it fails. Is there any solution to this problem?

collin80, 2020-12-16:
The solution is to either upgrade your version of QT or change that function call to swapAt instead. For some reason I may never understand, QT deprecated the old name so I changed the name in order to squash a warning about using the old name. But, the new name doesn't exist until a very recent version of QT.

tsunemitsu711, 2020-12-16:
Thank you for your reply.
Thanks to you, I was able to move it successfully.
It didn't work with 'swapAt', but it worked when I changed it to 'swap'.

closed 2020-12-16 by tsunemitsu711