collin80/SavvyCAN #272: byte order is displayed wrong for big endian signal

BitterAndReal, 2021-01-31:
If I uncheck the signal byte order in the DBC editing window the signal bits are highlighted different than I expected.
[image]

This CAN DBC editor playground marks the bits as I would expect for little and big endian.
https://docs.google.com/spreadsheets/d/1X6VQBkqRX0a6QLd4oFbAos_ps3zrpVc86CVGd67DoSs/edit?usp=sharing

collin80, 2021-02-01:
Yeah, start bit 15 and a length of 16 looks just like in your picture from the other editor. The part that's probably non-ideal is that SavvyCAN doesn't even attempt to change the starting bit when you click the checkbox on/off for little endian. It leaves the start bit alone. If you manually switch the start bit to 15 it looks correct. But, I'm inclined to leave it this way as the idea is that you intentionally set the start bit then edit the length and endian to match your signal. I think it might be weird if changing the checkbox also changed the start bit. But, maybe it could work.

BitterAndReal, 2021-02-01:
Sorry I'm still trying to get my head around it. 
I did not know the start bit for a big endian signal is the leftmost bit and counts down (to the right) in the byte before it jumps to the highest bit of the next byte.
I was expecting the same bits are used for a big and small endian signal and only the MSB and LSB would be inverted.
Very confusing and probably the reason little endian is mostly used for CAN signals.

closed 2021-02-01 by BitterAndReal