collin80/SavvyCAN #273: SavvyCAN installation issue

agbajepaul, 2021-02-03:
Please how can I solve this? After the instructions below

/opt/Qt5.9.9/5.9.9/gcc_64/bin/qmake CONFIG+=debug

make

...I am getting this error:

++ -c -pipe -g -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DQCUSTOMPLOT_USE_OPENGL -DQT_PRINTSUPPORT_LIB -DQT_HELP_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_QML_LIB -DQT_NETWORK_LIB -DQT_SERIALBUS_LIB -DQT_SERIALPORT_LIB -DQT_SQL_LIB -DQT_CORE_LIB -I. -I/opt/Qt5.9.9/5.9.9/gcc_64/include -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtPrintSupport -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtHelp -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtWidgets -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtGui -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtQml -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtNetwork -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtSerialBus -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtSerialPort -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtSql -I/opt/Qt5.9.9/5.9.9/gcc_64/include/QtCore -I. -isystem /usr/include/libdrm -I. -I/opt/Qt5.9.9/5.9.9/gcc_64/mkspecs/linux-g++ -o main.o main.cpp
In file included from mainwindow.h:21,
                 from main.cpp:1:
dbc/dbcmaineditor.h:8:10: fatal error: QRandomGenerator: No such file or directory
    8 | #include <QRandomGenerator>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

collin80, 2021-02-03:
You have to use a newer version of QT. QRandomGenerator was added in QT 5.10 I think. The documentation for SavvyCAN might be a bit out of date. Yes, it says you need QT 5.8 but that's wrong. Using at least QT 5.12 is probably required. I compile with QT 5.15 at this point.

agbajepaul, 2021-02-04:
Thanks, this should be helpful.