collin80/SavvyCAN #275: Question on integration with QCanBus devices

mgsdk, 2021-02-12:
The current implementation of SavvyCAN allows for easy integration with devices, which are mapped to a QCanBus device class. I have two questions on how to properly map CAN messages from the device into the SavvyCAN interface:

1. How to best represent absolute timestamps from the device
2. How to represent CAN frames sent autonomously by the device

#### 1 - Timestamps

With regards to 1, what would be the best solution to map CAN frames from a device with an incorrect clock. E.g. if the device clock is in the past, device timestamps currently result in abnormal numbers due to an underflow at [serialbusconnection.cpp#L238](https://github.com/collin80/SavvyCAN/blob/master/connections/serialbusconnection.cpp#L238), since the `timeBasis` is newer.
Would it be possible to have an option for keeping the device timestamp absolute, such that users can opt to set `timeBasis` to 0?

#### 2 - Marking a message as tx

The issue for part 2 is due to the user having the option to configure the device in advance, including setting up a series of messages to transmit at pre-determined intervals. It would be nice to expose this to the end user, since SavvyCAN already has support for distinguising between rx and tx CAN frames.
[serialbusconnection.cpp#L234](https://github.com/collin80/SavvyCAN/blob/master/connections/serialbusconnection.cpp#L234) seems to force all messages received through the QCanBus interface to be marked as rx. I do not have a huge insight in the other modules mapping CAN frames in through this interface, but perhaps you could sched some light on the possibility of utilizing the `QCanBusFrame::hasLocalEcho()` value as indicator for rx/tx of each frame?

I can provide PRs with candidates for both issues if desired?

collin80, 2021-02-14:
1: Yes, currently there needs to be some work put into making the timestamp calculations work better. It probably is a good idea to allow the device timestamp to be absolute with the timeBasis being 0.

2: Oh, I see what you mean, the code is locking all frames to be seen as "received" right at the driver level. Yes, it does look like hasLocalEcho is probably the proper solution here.

Thanks, I think both things would be beneficial to SavvyCAN.

mgsdk, 2021-02-15:
I probably need to tweak the GUI a bit to cover 1 in a sensible way, will post a PR when I have a candidate.

Part 2 is covered in #278 .

MatinF, 2021-02-18:
Hi Collin, for point 1 (on timestamps), would this be something you'd be open to modifying in the code? We looked through it, but we're a bit unsure on how to best adjust this to ensure it's also compatible with other integrations. 

The ability to show absolute timestamps would be nice, but a short-term fix could be to simply offset the timestamp from the integrated device in cases where that timestamp is less than the PC time. This would help avoid 'abnormal' timestamps in a simple way. Happy to hear your thoughts on this.