collin80/SavvyCAN #276: Support for 8Devices Korlan USB2Can logging (Win10)

VKTechnician, 2021-02-14:
Hello,

Was wondering if this could be added to the list of QT devices. The USB2CAN device works well with Wireshark, but Wireshark lacks DBC file and message filtering capabilities.

gedass2000, 2021-02-14:
It works with j2534 plugin

collin80, 2021-02-14:
As gedass2000 said, you can use any device that has a J2534 driver so that's an option. Do you know how Wireshark interfaces with it? Does it use J2534 as well? It's possible to write a driver specifically for it but we tend to shy away from writing drivers for every individual device if they can be supported via the more traditional channels. LINUX is great because nearly everything supports socketcan so writing a driver usually isn't necessary.

rusoku, 2021-02-14:
Soon I will publish on Rusoku github both J2534 and Qt plugin for usb2can and Rusoku TouCAN converters.Additionaly will be added Qt plugin with MacCAN driver support from uv-software for Rusoku converters series.

sresam89, 2022-04-02:
@rusoku I would like to contribute to the usb2can plugin as well, let me know if there is a PR I can start

rusoku, 2022-04-02:
@sresam89 what do you mean "usb2can plugin" ?

sresam89, 2022-04-03:
> Soon I will publish on Rusoku github both J2534 and Qt plugin for usb2can and Rusoku TouCAN converters.Additionaly will be added Qt plugin with MacCAN driver support from uv-software for Rusoku converters series.

I was referring to the QT plugin in your comment earlier.

I bought [this device ](http://wiki.inno-maker.com/display/HOMEPAGE/usb2can), it has a .dll file of its own It compatible with [Cangaroo](https://github.com/normaldotcom/cangaroo) which makes me think porting this .dll to QT supported canbus-devices would be easy.

gromgsxr, 2023-01-26:
what would need to be done to get it to work?

rusoku, 2023-01-26:
To develop Qt plugin

gromgsxr, 2023-01-26:
Something like this?  https://github.com/BugLight/qt-candle-plugin

gromgsxr, 2023-02-24:
> Something like this? https://github.com/BugLight/qt-candle-plugin

Is this not suitable?

rusoku, 2023-02-24:
With firmware upgrade only.I don't support Korlan and CANAL library anymore.
New iteration adapter is TouCAN and it use MacCAN and TouCAN API for Macos, Windows and Linux.
Qt 5.15.x plugins available as well.