collin80/SavvyCAN #280: segmentatioun fault when add new signal or message.

iiKelZenii, 2021-02-17:
I try to create my own DBC file. But if I try to create a signal it hard fault crashed.
This video linc shows a bug
https://yadi.sk/i/msRhnyMTU1_uIQ
Here is my information about the sitsem:

uname -a
Linux kelzen 5.4.0-26-generic #30-Ubuntu SMP Mon Apr 20 16:58:30 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 20.04 LTS
/home/kelzen/Qt/5.15.2/gcc_64/bin/qmake --version
QMake version 3.1
Using Qt version 5.15.2 in /home/kelzen/Qt/5.15.2/gcc_64/lib

P.S. Sorry, I'm bad at English.

collin80, 2021-02-18:
извините...

It should be fixed as of the commit I just did: bc84959093001484f8b1662c517422675f3aca00
There were a lot of stupid errors in the code that caused crashes. It should be more stable now.

iiKelZenii, 2021-02-18:
Thank you so much. Now it works. This is a great app.

closed 2021-02-18 by iiKelZenii