collin80/SavvyCAN #287: Linux build socketcan

jonvolk, 2021-02-21:
Im running slcan through a custom STM32F105 board and having some issues in reading CAN messages. If I turn on fuzzing it spews messages out on the bus, but if I just turn on capturing and try to receive messages it comes up empty.  Ive validated the STM firmware is receiving and sending over usb with 'candump' in a terminal.

Would the following be indicative of a "somewhat broken" Linux release or a more local error?

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
QObject::disconnect: Unexpected nullptr parameter
QObject::disconnect: Unexpected nullptr parameter
qt.canbus.plugins.socketcan: Cannot load library socketcan: (socketcan: cannot open shared object file: No such file or directory)
qt.canbus.plugins.socketcan: Cannot load library libsocketcan, some functionality will not be available.
Cannot load library socketcan: (socketcan: cannot open shared object file: No such file or directory)
qt.canbus: Can not disconnect an unconnected device.
qt.canbus.plugins.socketcan: Function can_set_bitrate() is not available.

collin80, 2021-02-21:
Yes, that certainly looks like something is broken with the socketcan support for QT. SavvyCAN uses the built-in socketcan support that comes with QT (I think QT higher than 5.8) but it appears that the QT plugin just can't find the proper socketcan libraries and so it doesn't work. But, unfortunately I don't really know why this is happening to you. You might need to reinstall both the socketcan tools and QT to make sure that everything is installed properly.

jonvolk, 2021-02-21:
Thanks. I'll try a fresh install/build and report back.

jonvolk, 2021-02-22:
That was an odd one. Reinstalled Qt and it now works both rx and tx. Oddly enough it still spits out the same error messages in the terminal..but it works and thats really all that matters. Thanks for the suggestion.

closed 2021-02-22 by jonvolk

kejava, 2021-12-30:
@jonvolk,
Sorry I'm late to this old topic, just started using SavvyCAN and ran into something similar. Figured it might be useful for someone new to this.
I'm running on MX Linux and found I had to install "libsocketcan2" deb package. That installed the shared lib into /usr/lib/x86_64-gnu

Even after that, the error persisted. I needed to make a sym link of "libsocketcan.so" for SavvyCAN (or Qt) find it.
`cd /usr/lib/x86_64-gnu`
`sudo ln -s libsocketcan.so.2.3.0 libsocketcan.so`