collin80/SavvyCAN #293: Read DBC file error decoding some char

manux81, 2021-02-23:
I found a bug reading a dbc file about unit measure name :
SG_ MOB_TEMP_DC_H : 63|8@0+ (0.75,-40) [-40|150] "°C"  BMU

So I change bool DBCFile::loadFile(QString fileName) function to solve by my self in this way:
QByteArray rawLine;
QString line;
...
 while (!inFile->atEnd()) {
		rawLine = inFile->readLine();
		line = "";
		QByteArray tmp = rawLine.simplified();
		for (int c = 0; c < tmp.size(); c++) {
			line += tmp[c];
		}

In this way, all chars are correctly copied to QString line.

cheers,
Manuele

kdschlosser, 2021-02-25:
do you have the file encoding for the DBC file set to "windows-1252" or "cp1252" the dbc file has to be saved with this encoding in order for it to be read correctly. It also has be be opened using this encoding as well.  This is the file encoding that Vector uses,

manux81, 2021-04-03:
HI @kdschlosser
the file is attached 
[example_dbc.txt](https://github.com/collin80/SavvyCAN/files/6252596/example_dbc.txt)

collin80, 2021-04-04:
I looked at your file and, sure enough, the units in question do show up oddly when I load it in a text editor. It does appear to be simply a problem with the file encoding. Apparently QT is correcting that encoding issue with the code you posted. kdschlosser says that CP1252 is the proper formatting but I'd imagine that UTF-8, which is the QT5 default, would work fine as well.

kdschlosser, 2021-04-06:
The file that is attached above is encoded using ANSI. this would explain why there is a problem when trying to load the file with Savvy CAN.

the file cannot be converted to CP1252, it can be converted to UTF-8

I do know that if a file is opened in Vector CANdb++ and the file is encoded with UTF-8 the characters will not display properly. The file MUST be encoded with CP1252 in order to open  properly in CANdb++

closed 2022-12-09 by collin80