collin80/SavvyCAN #300: Feature: Add 'size to plot' in the Graph tool

MatinF, 2021-02-24:
Hi Collin,

When working with data from e.g. log files in SavvyCAN, the Graph tool would be a great supporting tool.

However, when creating a new plot it often takes quite some effort to navigate to the chart. It would be really useful to introduce some of the below enhancements:

1) Add a size to plot via the right-click, which sets the time and Y axis so that the plot(s) takes up e.g. 80% of the window
2) Perhaps "cap" the amount that a user can "navigate away" from the data - currently I often get lost and need to start over. For example if the max data point is 800, the Y axis could be max capped at 1200.
3) For multiple plots, maybe an option to set one to the Y-axis would be useful - though this may be a more heavy effort feature

best, Martin

briantrice, 2021-03-14:
I agree that this would be helpful, and am making some proposed changes to address it.

MatinF, 2021-06-11:
I believe the latest versions have this feature as something that can be selected when right-clicking the chart, which is great. Perhaps one last step to close this would be to make it the "default" behavior, so that when a user adds a new signal, the plot will automatically zoom in on it (at least for the first signal added). This would greatly improve the UX as users won't be confused from not seeing their data. Happy to hear your thoughts :-)

closed 2023-03-24 by MatinF