collin80/SavvyCAN #301: Multiplexor after signal does not work

BitterAndReal, 2021-02-25:
The multiplexed signals are not recognised if the multiplexor is after the multiplexed signal in the CAN message.
I set up a test where I send the similar message with two ID's. 
One has the multiplexor signal in the first two bytes and the other in the last two. The multiplexed signals are in byte 2-5.
0x064 with the multiplexor in the beginning works fine
0x065 with the multiplexor in the end is not interpreted
I added printscreens and the DBC file I used for the test.

[mux test DBC.zip](https://github.com/collin80/SavvyCAN/files/6043067/mux.test.DBC.zip)

[image]
[image]
[image]
[image]

collin80, 2021-02-25:
I'll check this out. However, I'm currently totally rewriting the DBC code to support extended multiplexing. I believe my code changes will fix this problem as well.

BitterAndReal, 2021-02-25:
That's nice! I like your work very much.

collin80, 2021-04-04:
The code was checked in for extended multiplexing and I'd think it probably fixed this issue as well. So, if you get a chance, you might try again and see if things are working better. If so we can close this issue.

BitterAndReal, 2021-04-05:
Thanks! I'll have a look...

BitterAndReal, 2021-04-22:
I tested with the version 1.0.173 released on 19 Mar but I expierience the same problem.  
[image]

collin80, 2021-04-29:
Yes, apparently there was a bug that would not allow multiplexors to work if they are big endian and at the end of the frame. I've corrected that but haven't merged in the change to github yet. But, it should drop later today and then it should work both at the beginning and end.

BitterAndReal, 2021-05-18:
Multiplexing is working perfect now. Thanks a lot!

closed 2021-05-18 by BitterAndReal