collin80/SavvyCAN #305: Error while trying to compile for Linux

mharrow, 2021-03-01:
../Qt/5.15.2/gcc_64/include/QtGui/qopengl.h:141:13: fatal error: GL/gl.h: No such file or directory
 #   include <GL/gl.h>

Do you have any idea why this error happens. Using Linux Mint 19.3 and installed Qt 5.15.2. Initial error pointed g++ not found. Install g++ with apt then the above error occured.

collin80, 2021-03-01:
Yes. Quite often people find that there are more prerequisite packages than they had assumed. It happens to me whenever I have to reinstall linux. Here's one resource that talks about your issue in particular:

https://wiki.qt.io/Install_Qt_5_on_Ubuntu

Basically, you need the MESA development library to get OpenGL support. QT uses OpenGL for some things so you need it to compile apps. This is especially true of SavvyCAN which can use OpenGL for graphing acceleration and anti-aliasing.

mharrow, 2021-03-01:
sudo apt install mesa-common-dev libglu1-mesa-dev

Thanks Collin. This fixed the problem

closed 2021-03-18 by collin80