collin80/SavvyCAN #320: Save Spreadsheet Data shows random numbers in excel.

Breastfed, 2021-03-18:
Hi 
I have several parameters of engine data captured from 1000 to 2500 RPM, When I view the Graph data everything looks consistent within this range meaning no spikes or negative values, however when I "Save Spreadsheet of data" to xxx.csv file and view the file I have these random numbers in red that are unrealistic and definitely not visible in the Graph. I would guess that as much as 20% of the data is like this.

Any thoughts
Breastfed

[image]

collin80, 2021-03-18:
Unfortunately I did know about this issue already. It seems that there are various conditions that can cause the spreadsheet output to really foul up badly. I thought for sure that there was already an issue for this but I can't find it. If I remember correctly, part of what generally triggers this is if your capture is not strictly in time order. If the timestamp was reset in the middle then weird things happen. I believe the solution is to sort the capture before generating the spreadsheet.

I'll leave this one open and active until I fix it properly.

Breastfed, 2021-03-19:
Hi Collin 
What do you mean by Time Stamp was reset in the middle ? 
FYI the log files size is set to 20MB this is about 20minutes worth of data when logged at the max rate or every 1ms ... 

Success it seems, Today I reconfigured the Data logger to "down sample" at a rate of 20 (every100ms) this helped a lot as I only had 6 lines with funky random data.

Breastfed

collin80, 2021-03-19:
Let's say you have 5 entries. If the time stamps are 1,2,3,4,5 then it should work fine. But, if the stamps aren't continuously incrementing you run into trouble. Say, if the stamps are 5,1,2,3,2 or something like that. Currently it assumes the timestamps are perfectly in order and ascending. It probably won't tolerate two adjacent timestamps being the same either.

So, if there are any glitches in the timestamps it can do weird things. I don't know if you might possibly be experiencing this issue or if it is something else. But, past experience has shown that timestamp quirks have led to trouble.

Breastfed, 2021-03-29:
Hi
I just logged 3 hours of data on Sunday after exporting the graph to csv i would say that around 40% of the 578,428 lines consist of random number as discussed earlier
The original file only had 5116 duplicate time stamps

Can you able to suggest a work around for this issue ?

Breastfed

collin80, 2021-04-01:
Well, the work around is to not have timestamps that don't go in order but apparently the system you're using does generate duplicates or wrap around values. 

But, really I have to fix up the code so that it more gracefully deals with weird time stamps. There is currently "normalize time stamps" button in SavvyCAN. I could make it really normalize the timestamps and fix wrap around values.

PR #436 (2021-12-05): GraphingWindow: improve interpolation for CSV-export.
Fixes #320.
Note: Interpolation is still not quite correct and may have remaining artifacts at the epsilon-T region.
commit aa449d7e (2021-12-05): GraphingWindow: improve interpolation for CSV-export.
Fixes #320.

Note: Interpolation is still not quite correct and may have remaining artifacts at the epsilon-T region.
closed 2021-12-05 by collin80