collin80/SavvyCAN #323: IDEA: Can Fuzzing condition met input

jamiejones85, 2021-03-29:
When can fuzzing, for example, I have a contactor control board and I'm fuzzing to try close the contactors.

When I send anything with the id 0x0BA it seems to wake up and start spitting out messages, so now fuzzing bits on this ID and listening for a click.

It would be cool if an input on the Due could be used as a success flag so it can be left running until this pin goes high, SavvyCan could then stop fuzzing and flag the message where the condition occurred.

collin80, 2021-04-04:
Early on in this hardware and software journey I did plan for such a thing. The GVRET firmware does have support for digital I/O. The idea was to be able to do things like you suggest - trigger an output when a certain condition is met or watch a digital pin and give a flag when that pin changes state. So, I meant to do this. Alas, it never really happened. The code is still in GVRET to do that but SavvyCAN never really got any capabilities to interact with that digital I/O system. Also, the subsequent versions of GVRET for newer boards like the ESP32 board don't retain enough code for the digital I/O stuff to work any longer. So, I meant to support this but... whoops, I didn't. However, I still do think it'd be a cool idea. If I ever get any time I would like to do this. Alternatively, if anyone else is reading this and would love to do it, by all means!

jamiejones85, 2021-04-06:
`case PROTO_DIG_INPUTS:
                //immediately return the data for digital inputs
                temp8 = getDigital(0) + (getDigital(1) << 1) + (getDigital(2) << 2) + (getDigital(3) << 3);
                buff[0] = 0xF1;
                buff[1] = 2; //digital inputs
                buff[2] = temp8;
                temp8 = checksumCalc(buff, 2);
                buff[3] = temp8;
                SerialUSB.write(buff, 4);
                state = IDLE;
                break;
`

I think this is what you're talking about, 0xF1 0x02 will contain the GPIO states?

And in SavvyCan

`case GET_DIG_INPUTS: //get two bytes. One for digital in status and one for checksum.
        switch (rx_step)
        {
        case 0:
            break;
        case 1:
            rx_state = IDLE;
            break;
        }
        rx_step++;
        break;
`
So the above would need implementing to actually do something with the data?