collin80/SavvyCAN #325: ISOTP-Decoder values incorrect

mickeyl, 2021-04-07:
On macOS 11 w/ QT 5.15, I found a problem in the values of a capture UDS communication, where the values of p2max and p2*max are much too high (512ms and 51200ms shown). After setting a breakpoint in `uds_handler.cpp:380`, it looked like a problem with

```c++
    const unsigned char *data = reinterpret_cast<const unsigned char *>(msg.payload().constData());
```

For some reason this leads to `data` pointing to ```0x8002001400c800``` instead of the expected
```0x021400c800``` – which then makes the computation

```c++
                int p2 = data[1] * 256 + data[2];
```

take the wrong values. Any idea what could cause this offset?

collin80, 2021-04-29:
Well, that's an odd one. I can see that it did in fact leave the pointer alone except that it added 0x80 to the beginning of the address. I wonder if this added offset means something to MacOS?! That new address is WAY up there in virtual memory. It must mean something but I've been unable to figure out what. I suppose I might be able to work around this but I'm pretty sure this does not happen on Windows or Linux.

Alex-Markov, 2022-10-23:
This does happen on Windows version V207. See below.

Raw Payload: 0x50 0x03 0x00 0x32 0x01 0xF4 

UDS Positive Response
Service: Diagnostic session control
Session Request: Extended Diagnostics Session
P2MAX (Max Wait / Resp Time): 768ms
P2 Ext MAX: 128010ms

Alex-Markov, 2024-11-16:
Original code for P2max
`p2 = data[1] * 256 + data[2];`
for P2 Ext max
`p2 = (data[3] * 256 + data[4]) * 10`
data[0] and data[1] are actually the service ID (SID) and subfunction code. Correct payload starts with data[2]. So the below should be correct.
for P2max
`p2 = data[2] * 256 + data[3];`
for P2 Ext max
`p2 = (data[4] * 256 + data[5]) * 10`