collin80/SavvyCAN #330: Extended 29 bit ID's miss the extended flag

BitterAndReal, 2021-04-22:
0x18FFFFFF is simply stored as decimal value 419430399 but it should be 2566914047.
The DBC ID adds 3 extra bits for 29 bit CAN IDs to serve as an extended ID flag.
Is there a way I can change the message to extended ID?

collin80, 2021-04-29:
I've added code to theoretically handle both this and the value tables. But, it's not a great implementation yet - basically it now allows the bit 31 to be set and just strips it off everywhere. But, upon saving if the ID is above 0x7FF it will automatically be set extended and then bit 31 set. So, it will "sort of" do something close to correct but not actually correct. A correct implementation is still to be done so I'm not closing either issue yet. But, this should temporarily make it work.

BitterAndReal, 2021-04-29:
Thanks for the temporary fix anyway!

BitterAndReal, 2021-05-18:
I tested this but I could not create a DBC with extended ID inside SavvyCAN.

The following two ID's are both possible but not identical!
0x00000600 (29 bit ID) = DBC ID: 2147485184
0x600 (11 bit ID) = DBC ID: 1536

Loading a DBC with extended ID's does kind of work. SavvyCAN is not making a difference between standard and extended messages. Both the standart and the extended decoding is used on both messages.

Saving a DBC with extended ID's does not work. All ID's are saved as standard ID's!
Bit 31 is also not set above 0x7FF!

This is a DBC file with a basic ID 0x066 and a extended ID 0x00000066 loaded with SavvyCAN:
[image]
[image]
[image]