collin80/SavvyCAN #350: HTML help files not up-to-date with latest fixes on the RST files

jogo-, 2021-05-20:
The HTML help files were not re-built at all to include all my latest fixes on the RST files.
Could you please manually trigger the documentation dedicated build task and/or eventually fix the main make ?

collin80, 2021-05-22:
I just now re-ran the build script. But, it looks like Sphinx is being irritable for me on Ubuntu 21.04 so it took a bit of work. I'll have to get that fixed for me and also add the sphinx documentation build to the CI scripts.

collin80, 2021-06-16:
And, about 2 weeks after that I converted all of the help documentation to github markup files that QT can directly load. This has improved compatibility. Also, it allows the help to be directly viewed on GitHub too.

closed 2021-06-16 by collin80