collin80/SavvyCAN #361: Suggestion: Simplify 'Frame Sender' and make it part of default view

MatinF, 2021-06-11:
**Challenge**
When using SavvyCAN for testing and simulation, it is very useful to be able to both monitor the 'transmit' window (Frame Sender) and the 'receive' window (the default trace view) at the same time. However, today, the transmit window is an overlay, which makes it more difficult to manage the two in parallel - in particular if the trace view has a lot of data. 

In addition, for users that are new to the Frame Sender logic it can be a bit difficult to understand the logic.

**Suggestion 1: Embed Frame Sender in default view for streaming**
To improve the user experience, my recommendation would be to make the following adjustments to the default view when opening SavvyCAN in 'streaming' mode:

1. Enable Overwrite mode by default when streaming data (but not when loading log files)
2. Make the Frame Sender window an integrated part of the default view as in below illustration - and make it an "embedded" window instead of a movable overlay. The height of the Frame Sender section could still be altered (if more/less space is required)
3. Users can still "exit" the Frame Sender window via an "x" in the corner

I think many of the other tools in SavvyCAN make sense to have as overlay windows as they typically do not have direct relations to the default trace view. But with the Frame Sender, you typically want to monitor a response from the CAN bus you're sending frames into (e.g. for OBD2, UDS etc purposes). Hence this change would be really helpful here.

[image]

**Suggestion 2: Simplify the Frame Sender fields**
Currently, it can be a bit difficult to know the required syntax for the various fields in the Frame Sender tab (e.g. dec vs hex, units etc). A way to simplify this might be as follows:

1. In the ID section, rename the column to `ID (HEX)` and enforce a HEX syntax. Optionally, a `0x` could be added automatically in front of the ID entered
2. For the En/Ext/Rem boxes, it is a bit unclear whether to click the box or write something next to it - suggest centering the check boxes to simplify this and remove any ability to write anything next to them (but maybe I misunderstand the logic of being able to write something in the fields next to the boxes)
3. In the data bytes section, I would suggest enforcing a HEX structure as well and calling the header `Data (HEX)`. Further, I would suggest splitting the data field into 8 separate data fields with enforced HEX syntax. If the purpose of the variable length today is to allow for CAN FD frames, maybe make CAN FD a separate "checkmark" that can be toggled, in which case the data field is not split, but instead functions more similar to how it does today. 
4. In the Trigger section, maybe rename the header to `Trigger (ms)` and avoid adding the unit explicitly in the field
5. Across all fields, I would recommend blocking invalid syntax - e.g. ensuring that the Len field can only contain integers etc

The above are just some suggestions on how to potentially make the Frame Sender more user friendly for new users. If I've missed some nuances, let me know. Happy to hear your thoughts!

collin80, 2021-06-24:
Hmm, yes, I have seen other programs where the sender is part of the main screen in some way. It does make sense that one would want to see both at the same time. Currently you could do this manually sort of like you did where you place the windows on the screen with one on top and the other below it. 

I'm not opposed to the idea of integrating it into the main screen, potentially in a simpler format. As it stands, the frame sender window is hopelessly complicated. You can do crazy things like trigger on another frame coming in then delay by a certain number of milliseconds all while grabbing bytes from incoming frames and modifying them. Is it powerful? Potentially. But, it's also confusing. I wrote it and I don't really remember the syntax for how to actually use it. I basically only ever use it for fairly simple stuff like sending a static set of data every X milliseconds. The frame sender was coded pretty early in the project (believe it or not...), well before the scripting interface. These days, I think anything too complicated should be done with scripts. That may allow for the frame sender to be simplified as you described and integrated as a sub-window to the main window like you have shown.

I'm fine with forcing it to be in hex for simplicity on the ID and data bytes. Most other areas in the program are always in hex too. I never realized that you could type things into the en, ext, rem fields! It was meant only to have check boxes. I didn't realize I had left text fields for those too. Whoops. I really should fix that.

I wonder if I should make the trigger only milliseconds for simplicity. Right now you can trigger by milliseconds, incoming traffic, and limit the number of times it triggers. That's kind of complicated and requires you to remember the syntax. Just putting a millisecond value in there would be easier. Then there's the modifications field - another complicated thing. You can change data bytes to do things like make simple counters. I feel like the syntax is overly rigid and hard to remember. If I kept these sorts of things I think I ought to instead make a visual editor system where triggers and modifications are done by selecting options from a graphical interface. Of course, that's a lot more coding.

Maybe it really is best to do one of two things:

1. Modify the current "custom" frame sender to instead exist in the bottom of the main window when you open it. Rename some things a bit, remove the text fields in the check box columns. Basically both fix it and make it a sub-window instead of it's own floating window.

2. Leave the custom sender alone but add a very much simpler frame sending area to the main window that is separate. This one would only allow sending on millisecond triggering, no modifications, just a simple frame sender.

One thing I don't want to do is make overwrite mode the default ever. I just don't like that frames would then be thrown away by default. I think I feel more comfortable capturing and saving all frames unless someone explicitly asks otherwise. But, this is a minor detail. Well, maybe I could add a preference where it would default to overwrite when streaming traffic if you have that preference option checked. I would NOT check it by default but if your workflow would improve with that default it could be optionally possible that way.

MatinF, 2021-06-24:
Thanks for your detailed reflections Collin, a lot of it makes sense. Below are my thoughts and suggestions:

1) I believe the advanced functionality of the Frame Sender tool is quite useful for some applications, e.g. if the purpose is to simulate changing CAN bus data then it's very neat to be able to set up the simple "functional logic"

2) At the same time, there is quite a lot of syntax to be remembered in the current version and very few "restrictions" on the input fields, which I think will add to the potential confusion for new users

My suggestions based on your comments would basically be your option 1 with a few tweaks:

A) Retain the current Frame Sender tool functionality, but simplify it and help "guide" the user more:
- Restrict input values for the "bus" field to be positive integers
- Enforce hex syntax for CAN IDs and Data Bytes (+restrict input values accordingly)
- Remove the text fields in the checkbox columns (and center the boxes)
- Simplify the Trigger to be purely in milliseconds (+restrict input values accordingly)
- Consider removing the DLC/length field, and instead purely let this be defined from the entry of data bytes
- Keep the modifications field

B) With the above simplifications, I think the frame sender could be "simple enough" to be integrated in the main window, taking up 30% of the height (ideally the height would be adjustable). Ideally it would still also be closable via an "x" in the corner. I think it would be nice to have the window available as a default window when opening up, but that is not critical.

Let me know your thoughts :-) thanks again for a great tool!

MatinF, 2023-03-24:
Hi Collin, I discussed this topic in our team as we have received a lot of feedback on the Custom Frame Sender from our users.

I think the Custom Frame Sender is one of the most versatile functionalities of SavvyCAN. At the same time, as we discussed, I think there are some things that could make it even better than it is today.

To make this more practical, I have tried to condense the suggestions in this thread into concrete changes below:

1. Make the frame sender an integrated part of the main window (taking up ~30% of the height, placed at the bottom)
2. Change the `Bus` field to be a positive integer (> 0)
3. Change `ID` to `ID (hex)` and enforce hex syntax, `0x`. Restrict input values to be `0x0` to `0x7FF` for 11-bit CAN IDs and `0x0` to `0x1FFFFFFF` for 29-bit CAN IDs. If a user enters e.g. `123`, it should automatically be changed to `0x123` in the field 
5. Change `En` and `Ext` by removing the 'text fields' and centering the check boxes
6. Remove the `Len` field (let this be deducted from the payload length instead to avoid inconsistencies where a user e.g. has entered a length of 2, but added payload with 4 bytes)
7. Remove the `Rem` field (I am not sure what this does)
8. Change the `Data` to `Data (hex)` and enforce hex syntax in the format of `0xAA 0xBB 0xCC ...`. If a user enters e.g. `1234`, it should be auto converted to hex syntax as `0x12 0x34`.
9. Rename `Trigger` to `Cycle Time (ms)` and force it to be a positive integer value (> 0)
10. Make `Count` a read-only field (currently it can be selected as an editable text field) 
11. Keep the modifications field as-is
12. Avoid the current 'automatic' creation of a new row upon clicking 'enable'. Instead, make buttons for adding/removing selected rows at the bottom

I believe these changes would radically improve the usability of the frame sender and make it far more user-friendly. In my opinion, it would be a core change in taking SavvyCAN to the next level in terms of using it for professional use cases like testing, simulation etc.

MatinF, 2023-11-29:
Just wanted to bump this to say that I think these changes would help make the frame sender a lot more versatile and user-friendly. Hoping it'll be possible to implement in the coming future :-)

MatinF, 2024-05-29:
@collin80 thanks for incorporating some of the updates in the latest versions - in particular adding the frame sender window to the main view! I've updated the list from before with what I suggest as the main changes that would help make this even better:

The main purpose is to make the frame sender more user-friendly by restricting inputs and thus avoiding confusion, e.g. avoiding below scenario:
[image]

Below are the remaining changes I would recommend:

1. Change `En`, `Ext`, `Rem` by removing the 'text fields' and centering the check boxes
2. Force the `Bus` field to be a positive integer (> 0)
3. Change `ID` to `ID (hex)` + restrict values to `0x0` to `0x7FF` for 11-bit IDs and `0x0` to `0x1FFFFFFF` for 29-bit IDs
4. Change the `Data` to `Data (hex)` and enforce hex syntax in the format of `0xAA 0xBB 0xCC ...`. If a user enters e.g. `1234`, it should be invalidated - or auto converted to hex syntax as `0x12 0x34`.
5. Rename `Trigger` to `Cycle Time (ms)` and force it to be a positive integer value (> 0)
6. Visually, I suggest making the width expand to 100% of the frame sender window (currently it only uses 50% of the width on my PC)