collin80/SavvyCAN #382: Clear frames

exekutive, 2021-07-04:
It would be nice if the clear frames button did not also reset the ID filters, and the bus toggles.

exekutive, 2021-07-06:
I think the 'restart capturing' button was also resetting them. Kind of annoying.

exekutive, 2021-07-11:
Just to clarify, here is a real world example scenario ...

Let's say I'm working with a range of ID's I want to study.
Periodically, I may want to start with a blank log window.

1. I suspend the capture
2. I clear frames  to blank the window. **My ID filters selection is now lost.**
3. Now I must re-load my filter def'n list from file, so that only the IDs I want will be displayed when I resume.
4. I press "restart capturing", but now my filter list gets populated with new IDs that are **enabled** automatically, causing the window to fill up with frames I didn't want.

I think this is really basic stuff that is frustrating and needs to be fixed. My ID selection should **persist** until I change it **myself**. It's fine if the ID list is automatically populated, but selected IDs should not change, and none of the logging control buttons should affect my selection either.

collin80, 2021-07-11:
Definitely it is not proper for the restart capture button to reset the filters. As for the clear frames button doing it. Well, it seemed reasonable at the time. But, I can see how your use case makes sense. The idea was that if you clear the frames you start with a blank slate. But, it does mean you start with a blank slate. Perhaps what should really happen is that filters get a button to clear them explicitly and otherwise they don't. Well, perhaps it's still a good idea to clear filters when loading a capture file. In that case the new capture might have completely different or partially different frame IDs or otherwise be from a different source. So, maybe it'd still be OK to clear filters when loading a file. But, yes, I can see how clearing filters should be separate from clearing the frame cache. And, yes, any other logging control buttons shouldn't affect the filters either.

exekutive, 2021-07-12:
> Perhaps what should really happen is that filters get a button to clear them explicitly and otherwise they don't.

Exactly what I was thinking

> it's still a good idea to clear filters when loading a capture file.

That sounds logical