collin80/SavvyCAN #389: playback order

exekutive, 2021-07-06:
For some reason, Playback is playing the **first** frame from the log file, **last**.

[image]

[image]

[image]

NMSTEC, 2021-07-10:
Can confirm this bug on 500baud j1939 busses. 
However I have also seen the frame order being buggered up, and packets being dropped in listen only mode. Confirmed this by having M2 sniff side by side with an ixxat module/Can Runner. 

Stole quite a bit of my time from me. Narrowed it to savvycan, but not specifically where.

collin80, 2021-07-11:
Well... that's a... ah, special feature I guess. Going to have to fix that.

As for the frame reordering and dropping - could be the firmware on the M2 - M2RET. I'll start a new issue to keep track so I remember to look into it.