collin80/SavvyCAN #392: count field not saved

exekutive, 2021-07-11:
custom frame sender:
The "count" column is not saved when "Save Grid" button is used.

collin80, 2021-07-11:
Yes, at the moment the count column is used pretty much just for your reference to show that something did happen. It also is used if you ask it to send only X number of messages. Then it will stop when it counts up to that many. But, it isn't saved because it's only meant as a visual reference that sending has happened. I guess it could be saved but I hadn't considered any use case where it would seem necessary.

exekutive, 2021-07-12:
I haven't had use case yet either. But maybe someone could have a long frame list with carefully chosen values for number of times to send? I just brought it up in case it was an oversight. My feelings won't be hurt if you close this.