collin80/SavvyCAN #430: Connection not saved

mickeyl, 2021-11-09:
Is there a known cause for why the CAN connection is not saved? I'm pressing 'Save Connection', but without any obvious effect. Which file is it supposed to save the connection to? Perhaps there are permission problems.

collin80, 2021-11-10:
There is a setting in preferences (File -> Preferences from the main screen) to make it save/restore connections. It's off by default but you can click it on and it should start to work. The connections are saved with the system preferences system. On Linux that puts a text file in ~/.config/EVTV. On Windows I'm sure it's something similar. Since it uses the operating system's default way of saving settings it should not run into permission issues.

mickeyl, 2021-11-10:
Now I'm feeling dumb. I mistook "Save Bus Settings" in the connection window for the functionality that saves the connection. *phew*. Thanks for clarifying and sorry for the noise.

closed 2021-11-10 by mickeyl