collin80/SavvyCAN #435: [FeatureRequest] Differential timestamps

mickeyl, 2021-12-02:
One of the nice features, the console utility `candump` has, is being able to show timestamps as _differential_. Often, when comparing different flash tools, we need to inspect timing differences. With absolute timestamps, this is very involved.

collin80, 2021-12-05:
One of the newer commits (added as a pull request from someone else) has changed the sniffer window to have differential timestamps and also return the frequency in hertz for each message. I think this is more in line with what you were after. It appears there are some minor things I might want to change but it still should be fairly usable as-is.

mickeyl, 2021-12-05:
Amazing, thanks for the update. I will compile the latest version and have a look!

mickeyl, 2021-12-10:
This looks good. Could we get the same for the main window? And… if I may… perhaps also a way for the time base to be in milliseconds? Both seconds and microseconds are very tough to parse, in particular since many of the CAN timeouts are formulated in milliseconds (or even seconds).