collin80/SavvyCAN #440: Flow view displays garbage when DLC is less than 8

Astrinus, 2021-12-28:
[image]
[image]

In the log, message 0x101 has always DLC 1 and message 0x325 has always DLC 2, but SavvyCAN displays spurious bytes in the flow view.

collin80, 2021-12-29:
Yes, that's definitely not the way it should work. I'm guessing it just grabs random bytes from memory if they were not filled out. In reality it should grey out the bytes that don't / cannot exist for that message and also initialize them to zero so you don't see random garbage.

collin80, 2021-12-30:
I just pushed a commit that should mitigate this problem. It isn't 100% done properly yet (doesn't gray out data bytes past the valid point) but it does zero out such bytes so they don't show up as garbage. I will leave this open and try to get back to fixing it properly.