collin80/SavvyCAN #466: graphing window not respecting bus ID

daniel1c, 2022-04-04:
Hi Collin80, thanks for this great project!

I have multiple buses on a PEAK hardware logger, DBC assigned to bus IDs and trace window is interpreting the logged signals just fine.

Problem is now, that there is the same CAN std ID on two buses associated to different DBCs with different content. It appears, the plot function in graphing window does not check on which bus the message was captured. So the plot seems to be appended whenever the originally selected message ID shows up on any bus. 

Do I interpret this correct? Can a bus filter be implemented for the graphing function? Such thing seems to be working in trace window already.

I am using latest master, compiled with MinGW64.

Kind regards,
Daniel

collin80, 2022-04-16:
Yes, there are known problems with graphing and multiple buses. I do need to fix this. I will mark this as a valid bug.

collin80, 2022-12-09:
This was fixed/added in d6e7403d36bbdbc6491e8735db9b9c01195c8c90

closed 2022-12-09 by collin80