collin80/SavvyCAN #469: Needed: 32 bit version with 33kbaud (or custom baud) support

mazcoder, 2022-04-11:
Hey Collin.  Program looks awesome and I am looking forward to using with my old M2. Trying to run this on an older windows 8 tablet that is 32 bit. The SavvyCan32 version 199 functions decently but it  doesnt have 33k baud (which i think I need for my chevy volt). All versions on github appear to be 64. Is there anyway to do this? Do I need to compile myself somehow?

collin80, 2022-04-16:
You will probably have to compile it yourself. It shouldn't be too difficult. You just need QT and Visual Studio Express. I quit making 32 bit versions because 64 bit has been the default for a long time now. But, yes, there are still circumstances where it could be handy to have a 32 bit version. I'll see if it'd be easy to get the Ci/CD build scripts do a 32 bit version as well.

mazcoder, 2022-04-20:
I am fine to attempt to compile myself but QT appears to be a steep learning curve. I have installed QT 5.15 but I'm pretty sure it's not configured correctly and things that I was reading on the Internet seem to say that I need to have some sort of purchased license in order to configure it . Do I have to buy something in order to compile it?

rusoku, 2022-04-20:
I compiled win32 v207, Qt 5.15.2. Can you check it works or no because I don't have installed 32bit win version.
www.rusoku.com/downloads/SavyCAN_v207_win32.zip

mazcoder, 2023-01-17:
Sorry for my late reply. This project fell by the wayside for me. @rusoku THANK YOU SO MUCH! I verified this version you sent does in fact run well on x86 windows 8. And the 33k baud bus is there so this is issue is closed AFAIAC. Thank you