collin80/SavvyCAN #471: Connecting to socketcand

vekexasia, 2022-04-20:
hello, i'm trying to connect to a socketcand in my other network apparently discovery is not working but i was trying to use connection string.

I didnt seem to find the format so looking at the code it seemed to me that can://ip:port would work.

Unfortunately despite the fact that it says "connected" the console shows this:

```
Created TCP Socket to Kayak device can://ip:port
Reconnecting to TCP Host *ip*
Reconnecting to TCP Host *ip*
```
in loop.

If i try to do `telnet ip port` i get `<hi>` back.

QrackEE, 2023-06-11:
Hi,

Long time since your reporting, but I just got started with all the CAN tooling (and unfortunately need to find a way to work cross Linux/Windows platforms...). Anyway, today I found following combination string to be working so that it connects to non-broadcast-found sockets:

`<channel>@<hostname_string> (can://<ip>:<port>)`

where `<hostname_string>` can be basically anything - important part is to have IP:PORT in brackets.

My follow up later is that often I get 

> Created TCP Socket to Kayak device <channel>
Reconnecting to TCP Host <ip>
Opening CAN on Kayak Device!

with no following 

> Switching to rawmode...

Which makes my `socketcand` daemon (given random Fuzzer data being generated)

> unknown command '< send 300 1 15 >'.

I plan to follow up on that, but don't hold your breath.

gamebeast2k, 2023-12-30:
Hi, the connect issue is still here. (Build: Dec 20 2023) trying under win11 and win10

Savvycan dont anser the < hi > from remote with < open can0 >

Wiresharklog: 
Remote send : < hi > (Data: 3c206869203e)
Savvycan : ... send nothing 
10sec later
Savvycan : timeout reconnect

Settings:
Socketcand (autodectected srv) can0@xxx (can://11.22.33.44:29536)

With (nc 11.22.33.44 29536) 
remote: < hi >
local: < open can0 >
remote: < ok >
local: < rawmode >
remote: < ok >
remote: < frame 5E1 1703956223.642968 7A2A0060FE000000 > .......
works!