collin80/SavvyCAN #473: 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'

danielkucera, 2022-04-26:
Hi,
I cannot run the latest AppImage build on my Ubuntu with Wayland, the error is:
```
$ ./SavvyCAN-11e3b3e-x86_64.AppImage 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

(SavvyCAN-11e3b3e-x86_64.AppImage:501810): GLib-GIO-ERROR **: 08:04:05.532: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'
Trace/breakpoint trap (core dumped)
```

Jepson2k, 2023-05-30:
Having the same issue as well. Ubuntu 22.04 with wayland. Version v208 and previous have been working for me, but I went to upgrade to v213 and I'm getting this issue.

collin80, 2023-05-31:
For some reason this happens with Wayland. It doesn't happen if you don't use Wayland. That could be a deal breaker for you.  I will register this as a bug because multiple people have had this happen. I don't know why it suddenly hates Wayland but I'll need to find out.

photex, 2024-01-22:
Ahoy!

Just tried to run this, also on Wayland and I still get the error. Since basically every major distro is now just running Wayland I think it's hard to avoid.

Does building from source provide a work around? Is there any mitigation available for the AppImage?

danielkucera, 2024-01-22:
I am using a version compiled from latest master. That works fine.

danielkucera, 2024-01-22:
I have just found a simillar issue here:
https://github.com/Ultimaker/Cura/issues/12776

So if you don't want to compile, this should work:
```
GDK_BACKEND=x11 ~/Apps/SavvyCAN-11e3b3e-x86_64.AppImage
```

photex, 2024-01-22:
Oh that works! :D Thanks @danielkucera

italocjs, 2025-05-03:
Hello! i managed to open it using your tip, but it does not find the usb device (peak can), any ideas?  works fine on windows