collin80/SavvyCAN #475: Add settings option to display absolute date & time

MatinF, 2022-05-05:
Hi Collin,

Some of our CL1000/CL2000 users utilize SavvyCAN to load log files that were recorded historically. 

For such use cases, it would be very beneficial to be able to directly display the date & time, rather than the epoch timestamp. I imagine this could be an option enabled in the settings and that it could potentially be a relatively simple change in the code.

While I realize this would perhaps be less performing for real-time streaming, the main use case would be for analyzing loaded log files in my opinion. It could be disabled by default.

Happy to hear your thoughts on this!

collin80, 2022-05-05:
So you're saying this would be different from the current option where you can set the frame timestamps to be displayed as "System clock"? Because, that mode does show the historical time if it was stored in the original capture file. But, I just now noticed that it seems to ignore the format string you have set. That's got to be fixed.

MatinF, 2022-05-06:
Hi Collin,

I tried e.g. with the attached log file. It is true that using the system clock seems to produce some form of absolute date & time, but it does not match up with the corresponding epoch time when I use an online epoch converter.

[0000308.TXT](https://github.com/collin80/SavvyCAN/files/8637800/0000308.TXT)

Bert-JanK, 2022-05-10:
Hi, 
The same issue is for the "Data Graph" if I set the settings to "System clock", it will show some kind of human readable date and time, in the list, which is unfortunately incorrect but in the graph it stays at the Epoch time. I have added a screenshot with the first line of the log file shown in notepad and in SavvyCan. Hope this will help you understanding the issue and find a solution. 
If you wish I can upload a screenshot with the results of the same file and settings in the "Data Graph". 

[image]

M3ngoose, 2023-01-16:
Hi All,

Freshie here. I was using SavvyCAN with CSS CL2000 and came across this issue too. Found here and thought I could follow and try to contribute.

I experimented with the log file by modifying the "Time", "Time Format" and Timestamp, and giving them different values. Here's what I found:

SavvyCAN takes 1970-01-01 00:00:00.000 as base reference?
It interprets the timestamp minutes and seconds correctly (Time format: 2). For example, if I set the date to the base (1970) and entered a single line of data with timestamp of 02:12.000 (mm:ss.kkk), SavvyCAN converts this to the number 72 but incorrectly placed the decimal and displayed absolute time as 00:00:00.072 (hh:mm:ss.kkk) (72ms).

Hope this helps with the debugging.

Thanks for the contents.

collin80, 2023-01-17:
Ok, I finally gave it a try and I see that the 308 file linked above yields a system clock time of 18-January-1970 which is quite a bit off from Dec of 2017 like the log file actually was taken on. I found that the majority of the problem seems to be that the timestamp was being stored as milliseconds since the epoch but the timestamp storage is really done in microseconds so I had to multiply all timestamps from the file by 1000 to turn ms to us. This yields a MUCH closer timestamp but still it's a little off. The problem is that the code was adding everything on each timestamp line to the starting time specified at the start of the file. This is incorrect. Only the difference should be added. I've done all of that and tested with the 308 file and it loads properly now. If you set the mode to "system clock" now you will get the actual clock reading when each frame was captured. It should be accurate now but I only tested the one 308 file linked above.

Commit ID: 202756c667e8a7ae1f33dac8c476a074ab89774c

Bert-Jan48, 2023-01-17:
Hi Collin,

I have, a part of, the file in my comment above attached to this message in case you want to try an other file.
I will have a try myself, but I'am afraid it will take some time. 

[0000006.txt](https://github.com/collin80/SavvyCAN/files/10432349/0000006.txt)

closed 2023-01-18 by collin80