collin80/SavvyCAN #478: crash when connecting to socketcand

martijndierckx, 2022-07-02:
When connecting to socketcand, the messages start flowing. I can see them appear behind the connection window.
But after a couple of seconds, the app crashes.

Using V207 on Macbook Pro M1 Pro

Crash log:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               SavvyCAN [35158]
Path:                  /Applications/SavvyCAN.app/Contents/MacOS/SavvyCAN
Identifier:            com.yourcompany.SavvyCAN
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-07-02 18:10:27.4621 +0200
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        E009E7FC-4672-021E-3AD4-44B4EF075233

Sleep/Wake UUID:       F2479EE3-1BD4-4B6E-9168-D192CC2D8F01

Time Awake Since Boot: 110000 seconds
Time Since Wake:       1617 seconds

System Integrity Protection: enabled

Crashed Thread:        7  QThread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000200000024004e -> 0x000000000024004e (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x000200000024004e
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [35158]

VM Region Info: 0x24004e is not in any region.  Bytes before following region: 4375392178
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104ef3000-105187000    [ 2640K] r-x/r-x SM=COW  ...acOS/SavvyCAN

Thread 0::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff89fa36940 ???
1   libsystem_kernel.dylib        	    0x7ff80f48097a mach_msg_trap + 10
2   libsystem_kernel.dylib        	    0x7ff80f480ce8 mach_msg + 56
3   CoreFoundation                	    0x7ff80f583540 __CFRunLoopServiceMachPort + 319
4   CoreFoundation                	    0x7ff80f581bd0 __CFRunLoopRun + 1276
5   CoreFoundation                	    0x7ff80f581014 CFRunLoopRunSpecific + 562
6   HIToolbox                     	    0x7ff8186955e6 RunCurrentEventLoopInMode + 292
7   HIToolbox                     	    0x7ff81869534a ReceiveNextEventCommon + 594
8   HIToolbox                     	    0x7ff8186950e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
9   AppKit                        	    0x7ff811ef71fd _DPSNextEvent + 927
10  AppKit                        	    0x7ff811ef58ba -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
11  AppKit                        	    0x7ff811ee7f69 -[NSApplication run] + 586
12  libqcocoa.dylib               	       0x111253c64 0x111218000 + 244836
13  QtCore                        	       0x111a09d07 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
14  QtCore                        	       0x111a0dea2 QCoreApplication::exec() + 130
15  SavvyCAN                      	       0x104ef9d04 main + 548
16  dyld                          	       0x20546851e start + 462

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7fff369c4 0x7ff7fff32000 + 18884
1   runtime                       	    0x7ff7fff44460 0x7ff7fff32000 + 74848
2   runtime                       	    0x7ff7fff45ed8 0x7ff7fff32000 + 81624
[... 357 lines cut ...]
      "deploymentId" : 240000389
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,3, BootROM 7459.121.3, proc 10:8:2 processors, 32 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Thunderbolt Ethernet, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

martijndierckx, 2022-07-02:
Just tried the same build on Windows 11, which also discovered the socketcand connection.
But when trying to connect, no messages are being shown like on the Mac build.

Socketcand in verbose mode shows repeating connection attempts every few seconds:
```
client connected
Connection terminated while waiting for command.
Closing client connection.
```

collin80, 2022-07-07:
I will look into this. The socketcand driver was submitted by someone else and I've never used it. It's possible that a subsequent change somewhere in code broke the driver but since I don't ever use it I did not notice.

dittoheadd, 2022-09-04:
Setup `socketcand` to rpi, and got same connection problem
```
client connected
Connection terminated while waiting for command.
Closing client connection.
```
```
Created TCP Socket to Kayak device can0
Created TCP Socket to Kayak device can0
Reconnecting to TCP Host 192.168.1.102
Reconnecting to TCP Host 192.168.1.102
Opening CAN on Kayak Device!
Opening CAN on Kayak Device!
Send data to 192.168.1.102:29536 -> 
Send data to 192.168.1.102:29536 -> 
```

AndyHuska, 2022-09-17:
Possibly related to issues I found while running on Linux and fixed: https://github.com/collin80/SavvyCAN/pull/495